prusa3d/Prusa-Link #535: Bed temperature not consistently on trace

darraghbr, 2022-10-06:
Since updating to 0.7.0.rc1 I have noticed that the bed temperature trace sometimes disappears

[image]

For reference I am running on a Pi Zero 2 W and this has been happening for a number of days through a number of restarts.

TojikCZ, 2022-10-06:
Hi, This is not a PrusaLink bug, the Connect side should be fixed and released Soon™ now in testing

darraghbr, 2022-10-06:
> Hi, This is not a PrusaLink bug, the Connect side should be fixed and released Soon™ now in testing

Ok thanks, is it worth transferring this to another repo or just closing it?

ondratu, 2022-10-07:
That repo is closed. Creating issues here is ok ;) Thanks!

ondratu, 2023-05-30:
This should be fixed.

closed 2023-05-30 by ondratu