prusa3d/Prusa-Link #540: `Extruder fan` becomes `Hotend fan`

3d-gussner, 2022-10-18:
With MK2.5/S,MK3/S/+ firmware FW 3.12.0 and newer the `Extruder fan` will be replaced with `Hotend fan`.
To ensure consistent terms used across all printers, firmwares, documentation, etc.

Please update PrusaLink to support both or depending on the version the correct term.

Change https://github.com/prusa3d/Prusa-Link/blob/master/prusa/link/printer_adapter/state_manager.py#L442 ???
https://github.com/prusa3d/Prusa-Link/search?q=extruder
https://github.com/prusa3d/Prusa-Link/search?q=fan_extruder

See https://github.com/prusa3d/Prusa-Firmware/pull/3694

TojikCZ, 2022-10-18:
Attempt to fix in #542

3d-gussner, 2022-10-19:
Please feel free to close it as soon the PR has been merged

closed 2022-11-30 by ondratu