prusa3d/Prusa-Link #576: Intermittent beeping.  Hard to determine cause.

Vilhjalmr26, 2022-11-22:
On a random interval, my printer gives me 3 fast beeps in a row. I was able to see the display as it happened once, and it had something to the effect of "Internet connection error" and then immediately switch to "error resolved" and then went away. Hence me not having the exact text for this issue.
From looking through all the logs, best I can find is the following lines:

`raspberrypi connect-printer[545]: ERROR: HTTPSConnectionPool(host='connect.prusa3d.com', port=443): Read timed out. (read timeout=30) {loop_step():724}`

It would be nice if there was a log file that specifically listed errors that are pushed as notification to the printer like this, so there is an easier way to determine the cause of random beeping.

Also, is there a way to disable alerts for a given error? As having random beeps throughout my day is disruptive, and it does not seem to be causing any usage problems.

Printer: i3 mk3s
PrusaLink HW: Pi Zero W direct attached
PrusaLink SW: 0.7.0.rc1

TojikCZ, 2022-11-22:
Hi, we're aware of this annoyance, the quick fix would be to make the alerts optional, so that's where I'll start. I'd also like to make the network errors cumulative, so we can alert when there is a real persistent issue, not just an intermittent error. The cause of these is still a mystery tho. Something just interrupts the connection when receiving data, but that can be a variety of things. With the focus being set firmly on cameras, it's hard to convince anyone to go look at this one.

mix579, 2022-11-22:
I wish the focus was less on cameras but more on bread-and-butter stuff like this one or others such as ability to select multiple files for file management. Of course, easy for me to say as I'd rather use my Wyze cams than the junk Raspi cams, and they work perfectly fine….

TojikCZ, 2022-11-23:
Yea well, a lot of people have a lot of different ideas of what is important to them. I am an employee, I am supposed to do what I'm told. So yes, we know about this and we will counter it as soon as we can. As for any other requests, please use the forum or the discord server, don't wanna go OT under issues

kolins-cz, 2022-11-26:
Hello,
I can confirm this issue. Before you fix this issue for good, can you please suggest some quick and dirty way to disable beeps? Either in Prusa-Link or even in Prusa Firmware
Thanks

TojikCZ, 2022-11-27:
Sorry, it's only possible in firmware. Go to Menu -> Settings -> Sound choose Silent. Sorry for the inconvenience.

ablackbu, 2023-01-10:
Let me know if there’s any information I can provide when you’re able to look at it. 

Question about the workaround. Is silent mode going to disable all beeps or simply these ones? Load, unload, and filament runout beeps are helpful so don’t really want to miss out on those. My assumption is all or nothing.

kolins-cz, 2023-01-10:
> Let me know if there’s any information I can provide when you’re able to look at it.
>
> Question about the workaround. Is silent mode going to disable all beeps or simply these ones? Load, unload, and [...]

Printer will be completely silent. I do miss useful beeps, but random beeping from PrusaLink was intolerable for me.

ablackbu, 2023-01-22:
If you keep a compatible device around (ex mobile) around I found the prusa connect push notifications helpful. Even gives you a 5 minute warning push for color change. 

It has 4 options: browser, slack, discord, and telegram. 

I’ve found the discord we hook one to work very well.

I couldn’t figure out how to do it telegram, the browser didn’t work well (it did say it wasn’t supported so no foul there), didn’t try slack yet.

It would be great for some info on this setup process. Happy to help provide the some for discord.

aderusha, 2023-03-09:
I just updated to rc3 and it's now triple-beeping at least once every couple minutes.  It's driving me crazy!  There's got to be some way to disable this....

TojikCZ, 2023-03-10:
Sh*t I forgot to make it shut up. The only way to do thus remains to switch the printer to silent 😣

TojikCZ, 2023-03-10:
[image]

aderusha, 2023-03-10:
That is the best possible response to a GH bug that I've ever seen.  I'll just put mine on silent and wait for the next release :D

TojikCZ, 2023-06-02:
yup, the release has it silenced, you can turn it back on tho. I'll close this if that's fine with you all

closed 2023-06-02 by TojikCZ