prusa3d/Prusa-Link #605: Outdated firmware warning during setup

kubik369, 2022-12-22:
Hi,

I have tried setting up PrusaLink on my Pi (just installing from the git with pip). During the setup wizard, after connecting the printer, it exclaimed that I have old firmware and I need to update. This was not the case as I am running on the newest firmware. After refreshing the page, it happily let me through :)

Here are the logs (I was running with `prusalink -f`, so the included file contains whatever was the program outputting).
[prusalink.log](https://github.com/prusa3d/Prusa-Link/files/10290930/prusalink.log)

TojikCZ, 2022-12-23:
Hi, I am afraid there is no grace period for the error state in the wizard, so if it grabs the error between the printer connecting and the value getting read, it will present you with such state. This is nonsense of course. I considered it a lower priority as the error clears eventually. I want to improve the error handling as a whole. I did not succeed the first time around and now there were more important things to focus on. Will fix eventually. Sorry

kubik369, 2022-12-23:
No need to apologise, I am just reporting it here in case it hadn't been caught yet :)