prusa3d/Prusa-Link #608: Bad X-Api-Key error when trying to upload from PrusaSlicer

darraghbr, 2022-12-30:
I just updated to RC2, was working fine with RC1. 

I thought 'I wonder if the API key has changed' so I tried to upload from PrusaSlicer and got this error message.

[image]

After checking in the connect.prusa3d.com/printer/blah-blah-blah/settings  page I saw the API key had changed, so copy and paste into PrusaSlicer, hit test and I got the same error message.

I then went into the local UI just in case (prusalink.local/#settings) and in there was another API key, but a different one this time. Again copy paste into PrusaSlicer, same error message. 

I've double checked the IP in PrusaSlicer and that has not changed, tried the hostname instead, same issues. 

This issue is a real killer for me as I used it as part of my normal workflow so any help would be appreciated!

Sarlochin, 2023-01-02:
Change the "Authorization Type" in prusa slicer from "API key" to "HTTP digest" to keep up your workflow

TojikCZ, 2023-01-03:
You can still generate an API key under PrusaLink -> Settings -> API Key -> Reset
But we'd suggest sticking to the digest authentication. It is real shame, no-one is developing these standard login pop-ups anymore, expecting everyone to javascript together their own thing. These digest ones are still MD5 which is considered not secure. But no new algorhithms are supported by all major browsers.

darraghbr, 2023-01-03:
I've tried it again today using the API key in PrusaLink and it all seems to be working now, so I will close this. 

I don't understand, this may well just be my issue.

closed 2023-01-03 by darraghbr