prusa3d/Prusa-Link #610: PrusaLink produces invalid JPEG images from cameras on MacOS

mkoistinen, 2022-12-31:
JPEG Images encoded from the `blob:http://...` URLs used by PrusaLink are invalid and will not open in Safari on MacOS (Ventura 13.0.1). The image file produced cannot even open in Photoshop, which complains: "invalid SOS JPEG marker (Ss, Se, Ah, or Al)" and refuses to open them.

Interestingly, it appears Chrome, Brave, Firefox browsers on the same machine _can_ open them; presumably because they distribute with their own version of the libjpeg library?

The issue is mentioned here (https://gitlab.com/libtiff/libtiff/-/issues/266).

I wonder if using a newer version of the libjpeg library in the PrusaLink image is the solution?

ddx32, 2023-01-03:
It seems this is bug is predominantly affecting Logitech cameras. Firefox is doing the best job at correcting the corrupt data, but that's far from ideal, obviously.

TojikCZ, 2023-01-03:
I have noticed this but I don't know the solution, is there any magic corruption detection and correction in libjpeg-turbo I could try? One of our cameras seemed like it sometimes forgot to send the last byte of the image. Very annoying

JimBim89, 2023-03-17:
Does this help you? Quite old but seems to be related!

https://github.com/libjpeg-turbo/libjpeg-turbo/issues/157

TojikCZ, 2023-03-17:
So far, we've seen just one camera behave weird. It was also a Logitech and its frames seemed to be missing a single byte at the end. Please share your camera models. Plonk some frames on here if you can, so I have something to play with. Thank you

JimBim89, 2023-03-17:
It seems that Logitech C-Models are effected (mine is C615) and that they utilize a proprietary extension to the Motion-JPEG format, in which H.264 data is attached to the various JPEG key and delta frames in the APP0 field as shown when inspecting the data of any frame "AVI1". They also occasionally fill the buffer with two SOI markers, which could cause problems.

The picture frame in the linked issue hat the "AVI1" tag, I don't think its related to a missing byte here.

moofie74, 2023-06-23:
Logitech C270
[image]

godfreja, 2023-11-01:
I am seeing this issue with a Logitech HD Webcam C525

hilsonp, 2025-12-14:
I just migrated from Octoprint to PrusaLink using a C270 camera connected to the raspberry pi 3b+.
I'm using macos ans ios.

- PrusaLink in a browser (Firefox) on my mac was displaying the image
- PrusaConnect in a browser (Firefox) on my mac was displaying the image
- PrusaConnect app on my iphone was not displaying the image
- PrusaConnect tab in PrusaSlicer on my mac was not showing the image

I lost a few hairs until I understood that it did not display on the ios app and in PrusaSlicer because of safari engine versus the Firefox engine :-(