prusa3d/Prusa-Link #612: Display material, print time, weight and length of filament

Eldenroot, 2023-01-03:
[image]

We can see these info on display, why not in PrusaLink?!

TojikCZ, 2023-01-03:
Hi, this is not an issue with PrusaLink for the MK3, IDK where else to put it tho, so I'll leave this open for now and ask around.

TojikCZ, 2023-01-04:
closing this as this repo is not tracked by anyone developing for the MINI

closed 2023-01-04 by TojikCZ

Eldenroot, 2023-01-04:
Where to report then?

TojikCZ, 2023-01-04:
Okay, I will post the link explicitly: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2878
This mention link leads to the new issue reported under the right repo.
[image]