prusa3d/Prusa-Link #613: API key not settable

jbebel, 2023-01-03:
Using 7.0 rc2, the API key can only be reset, not set. This is inconvenient if one has several instances of prusaslicer using an old previously set API key. Ideally, after a reinstall and the key is lost, you should be able to set the API key of prusalink to match the one set in all your prusaslicers. Unfortunately now, you have to go an edit every prusaslicer to match the new key.

jbebel, 2023-01-03:
I guess http digest is the future. If so, then what is the API key for?

ondratu, 2023-01-04:
You are right. HTTP Digest is the right way. API Key is for some 3rd party applications for backward compatibility only, so set that by user is undesirable.

Thank you

closed 2023-01-04 by ondratu

TojikCZ, 2023-01-04:
You still can generate it under PrusaLink -> Settings -> API Key -> Reset but don't, if you don't absolutely have to.