prusa3d/Prusa-Link #614: Camera auto-detected but not detected?

jbebel, 2023-01-03:
I have a Pi 0 W with a pi camera module 2 connected. Prusalink discovers this and shows me a "unicam" at /dev/video0 using the V4L2 driver and listed as "Not Linked". This cannot be removed using the trash can icon as the camera is "an auto-detected camera". So clearly it found this on its own. It's auto-detected. But if I click on it, I get the message "Camera is not detected. Do you want to detect it manually?" Which is odd given that it was auto-detected. If I click confirm, it says a camera control request has been sent, but then nothing seems to happen and it remains unlinked. This needs better messaging regarding what system is and isn't detecting the camera, and some kind of reply when the request to detect the camera is sent, whether success or failure.

TojikCZ, 2023-01-03:
Yup, the filter supposed to ignore the pi-camera v4l2 handles is clearly broken, sorry. Please ignore the "unicam" camera. I am still woring on Pi camera support as the first attempt was very broken.

mix579, 2023-01-12:
I was just going to ask exactly the same question. I updated printer firmware and Pi image and can't for the life of me figure out how to get a Pi camera to work.

hreality, 2023-02-21:
Same, dropped a arducam (5MP 1080p) (though connect is running on zero 2w) with the same result

jbebel, 2023-03-21:
I was able to briefly see the camera, although then it seems like prusalink stops responding shortly thereafter. I haven't enabled ssh yet to take a look at why or how it's hung up. According to the instructions, that kills kittens.

TojikCZ, 2023-03-21:
Weird, so you are on RC3 with a supported camera - not the Raspberry Pi Camera V3 Wide one, that one slipped and is not whitelisted. You saw an image but then it did not update ever? The default is 30s, so please make sure it's really dead.

TojikCZ, 2023-03-21:
Seems not camera related, and if it is, it's a different issue. Closing this one. Feel free to open a new one once you find something out.

closed 2023-03-21 by TojikCZ