prusa3d/Prusa-Link #629: SSID with apostrophe causes boot failure

cdbax, 2023-01-24:
I couldn't get my Raspberry Pi Zero 2 W with the latest PrusaLink image (0.7.0rc2) to boot, and eventually noticed a single-quoted string in the first-boot script trying to escape the apostrophe in the SSID name. Changed to another SSID without an apostrophe in the name, and everything worked.

I think this will need to use a double-quoted string for the escaping to work.

ondratu, 2023-01-27:
Thank you for your bug report. We will try to fix it, but I'm not sure how it is possible. We use wpa_supplicant, and it looks like problem on it's side or on RPI-Imager side.

TojikCZ, 2023-06-02:
This might have gotten fixed with this PR #760 is this still a problem?

closed 2023-09-08 by TojikCZ