prusa3d/Prusa-Link #636: Connecting more than 3 printers to a network

Lheld818, 2023-01-26:
Hi Everyone,

I have a farm of roughly 55 mk3s printers. We have 5 pi's set up with prusa connect / link. However, we have only been able to connect 3 max to our network (industrial grade). Whenever we turn off one of the three, another one connects to fill the void.

Does anyone know the root of this issue? Appreciate any help.

Thank you

Lowentwickler, 2023-01-26:
Hi!

I would start from network settings. IPs, MACs etc. Make sure they are unique on all Pi boards. For example, when you turn off one printer, and turn on another one, will they both have the same IP address?

Also, could you please specify, what exactly happens when you connect forth and fifth printer?

ondratu, 2023-01-26:
Please, where exactly the problem is? Is it possible to connect all printers to your network - i mean getting IP address, working PrusaLinks on that and problem is only in connection between printers and Connect? Or you can't get IP address for more than three printers?