prusa3d/Prusa-Link #640: Webcam autodetec turn on/off

3d-gussner, 2023-01-31:
I am using/testing PrusaLink on my Linux PC which has one webcam connected. I started prusalink while being on a google meet call and everything was fine, prusalink couldn't get the webcam.
But as soon I temporary paused/stopped the meet cam, prusalink "hijacked" my webcam and I couldn't use the webcam in meet anymore.

This is somehow annoying that i had to "kill" prusalink to get the webcam free for meet.

Please add an option/toggle to en/disable auto webcam connect.

And as I am thinking of using multiple webcams and multiple printers on one device it would be great if you could assign one webcam to one printer, another to different one and so on.

RPis are :unicorn: and hard to find at this moment but, there are plenty of other low power solutions like thin-clients, micro pcs and other SBCs whcih can run multiple Prusalink instances and handle these and the webcams.

TojikCZ, 2023-01-31:
Put this into /etc/prusalink/prusalink.ini
`[cameras]
auto_detect=False`

3d-gussner, 2023-02-22:
Not working with 0.7.0-rc2 still auto connects

closed 2023-03-10 by ondratu

3d-gussner, 2023-03-21:
Please reopen it as 
```
[cameras]
auto_detect=False
```

is still in 0.7.0rc3 ignored and it hijacks my webcam.

TojikCZ, 2023-06-02:
This should be fixed in 0.7.0. Can I close this?

PR #684 (2023-03-10): Fix disabling camera auto-detect
Attempt to fix #640