prusa3d/Prusa-Link #655: Possible wrong "Printer" label

mmoraes80, 2023-02-10:
Hello,

After installing PrusaLink 0.7.0RC2, I noticed that the gray "Printer" label at the top of the page is currently showing the printer location instead of the printer's name.

Is this intentional or is it a minor code bug?

tomashusakk, 2023-02-13:
[image]
[image]

Hello,

Do you mean this printer label? There's a combined name of the printer and the location, as shown in the settings in the second screenshot.

mmoraes80, 2023-02-13:
Yes!. apparently it was a glitch with my browser. When I refreshed the page, it fixed itself.

closed 2023-02-13 by mmoraes80