prusa3d/Prusa-Link #692: 'Update is unavailable to install' message on 0.7.0rc3

darraghbr, 2023-03-10:
Updated to 0.7.0rc3 yesterday and when I go into the local address of the printer, go to settings and hit 'Check Updates' after a few seconds I am greeted with this in the bottom right.

[image]

Does this mean there is an available update and I cannot install it for some reason? Or does it mean there is no update available?

Might want to adjust the wording in order to make it clearer. For instance if everything is up to date a simple 'Prusa Link is up to date' may be sufficient.

TojikCZ, 2023-03-10:
This is a feature, the PrusaLink package does not exist yet, so checking for an update fails. Should fix itself once we actually publish it. Sorry for the error message

darraghbr, 2023-03-10:
I thought that might have been the case, thanks for the response.

closed 2023-03-10 by darraghbr

TheCase, 2023-05-27:
From what I can tell, the package is now available via pip, yet I still get the same error as above.

darraghbr, 2023-05-27:
Apparently the updater is broken in the release candidate and they are not sure how stable it will be in the full release version, more info here:

#758