prusa3d/Prusa-Link #695: Improved documentation for non RPI-zero devices.

MartijnGevaert, 2023-03-14:
Currently all the Prusa-Link documentation focuses on using the RPI zero, which are impossible to find at this time.
However, the README of this repo mentions "A connection over USB is also possible, making PrusaLink compatible with pretty much any Linux system".
I think it's really important to add more documentation on this. Believing that a RPI zero is required for Prusa-link was all I needed to know to start using Octoprint instead. It doesn't matter how good the software is if you can't run it.
It took visiting this repository and reading the README to find out that prusa-link also works over USB.
This opens up Prusa-link not only to all the other RPI's, but also any other linux device.

In contrast, it's very easy to find out that Octoprint can run on almost any device, even on an Android device. This simple lack of documentation makes Octoprint a much easier option to use with a Prusa MK3S, despite Octoprint having higher system requirements.
- https://github.com/feelfreelinux/octo4a
- https://octoprint.org/download/

mix579, 2023-03-14:
Agreed. Especially since running PrusaConnect over USB is trivial, I do it for all my printers using Pi 3 and 4s

MartijnGevaert, 2023-03-14:
It would be interesting to see what the system requirements actually are. For Octoprint if you go too low in specs it can affect print quality, but that shouldn't really be the case for Prusa-Link.
For Prusa-Link you might be able to get away with something like this: (https://pine64.com/product/128mb-ox64), but we don't really know what the ram requirements are.

The docs for prusa-link also says "Both will work, but Raspberry Pi Zero 2 W is recommended as it is faster than its predecessor, the Pi Zero W.". Which is certainly true, but it won't affect print quality. Might be better if this comment is removed, allowing people to save a bit of money.

ondratu, 2023-03-15:
Hi, yes, we are know about it. And actually we are working on documentation now. Please stay tuned.

MartijnGevaert, 2023-03-15:
Cool, feel free to close this ticket whenever you want.

Lukas-Heiligenbrunner, 2023-03-23:
Its also important to know that one have to disable the 'RPI support' in the printer settings for prusa-link to work over usb. Took me like 2 hours of debugging. ;)

TojikCZ, 2023-09-08:
This guide is now out https://help.prusa3d.com/guide/prusalink-prusa-connect-with-rpi-3-4-usb-mk2-5-s-mk3-s-_469341
Can we close this?

closed 2023-10-11 by TojikCZ