prusa3d/Prusa-Link #720: Non-OS package

CommanderRedYT, 2023-04-07:
Is there a way that you could make a non-OS-image version of this project? Then, we could install it ontop of existing Pis just like Octoprint. (they for example have also a Docker image)

This would mean that we do not have to use a Pi 4 for only Prusa-Link, but also for other things!

TojikCZ, 2023-04-07:
Hi, we probably won't provide packages for each linux distribution. I don't know about docker, but docker images are OS images, as far as my understanding goes. 

The readme contains a how to install guide. How is that worse than a docker image? You can create whatever setup you like using that info and you can put anything alongside PrusaLink once you install the image as well. Am I not seeing something?

CommanderRedYT, 2023-04-09:
Sorry, my fault, I somehow overlooked the pip package. :sweat_smile:

closed 2023-04-09 by CommanderRedYT

TojikCZ, 2023-04-09:
It is planned, it does not work yet. But the internal updater shipped in rc3 counts on it appearing next release 😅