prusa3d/Prusa-Link #729: "Upload to Queue" in PrusaSlicer uploads but does not enqueue it

davidstosik, 2023-04-18:
Hi there, it's me again! 👋🏻 
I'm not sure whether this is a bug in PrusaSlicer, PrusaLink or PrusaConnect, so apologies if this is the wrong place.

Here's a description of the steps to reproduce my issue:

1. Have a Prusa i3 MK3S (firmware 3.12.2-5713) printer setup with PrusaLink (0.7.0rc3).
2. Have that PrusaLink instance set up as a physical printer in PrusaSlicer(2.6.0-alpha6).
3. Also have that PrusaLink instance declared in PrusaConnect.
    (I have gone through multiple prints with this whole setup, so can confirm it works.)
4. Slice a model in PrusaSlicer.
5. Click on the bottom-right `G⏵` button to "Send to printer".
6. Click on "Upload to Queue".

**Result**: the file can be found in PrusaLink's storage, but was not enqueued in PrusaConnect's queue.

**Expected**: the file is stored in PrusaLink, _then_ put in PrusaConnect's queue. If the printer is ready and the queue empty, then I imagine the print would also start without needing further action.

TojikCZ, 2023-10-11:
Hi, sorry I missed this, is this still an issue? Does it happen for every model or just some of them?

davidstosik, 2024-01-01:
Sorry for my late reply! Having updated all versions, it looked like this is fixed now. Thank you!

closed 2024-01-01 by davidstosik