prusa3d/Prusa-Link #733: Short pause on "No internet access"

MartijnGevaert, 2023-04-18:
I am getting frequent beeps and errors as described in this issue:
https://github.com/prusa3d/Prusa-Link/issues/728

However, in my case my Rpi Zero is in a location with unreliable WIFI connection, so the warnings could be valid. However, it leads to short stops in printing (1~2seconds) after some of these beeps. These obviously lead to small blobs and surface imperfections.
The behavior looks like there is a small pause in GCode processing whenever the error is shown. I suspect this because the printer always continues printing without pause when a beep goes of and it's moving in a straight line. It does pause when a beep goes off and it's going over an area with a complex structure.

cqeta1564, 2023-04-23:
Hi, I have completely same issue like you. The best option is to use sd card for the gcodes. I hope that it will be fixed soon. Btw I use PrusaLink 0.7.0rc3.

davidstosik, 2023-04-24:
Related: #728

ondratu, 2023-04-28:
We had some problems on our Connect backend. That should be fine for now, but there is bug with doing blobs while Connect connection has troubles. So i close this issue after fix that bug.

closed 2023-06-02 by TojikCZ