prusa3d/Prusa-Link #737: Feature request: Flip/Rotate Pi camera image

mix579, 2023-04-23:
The way the camera is mounted on my Mk3S printers, the picture from the Pi camera in PrusaLink is upside down. Would love to see an option in the Camera settings to flip it/rotate 180 degrees. I suspect it's possible to do that on the Pi somehow but I know nothing about it, and shouldn't have to.

davidstosik, 2023-04-24:
I looked for it but did not find anything. In the end, I realized the new [camera mount](https://www.printables.com/model/46866-prusa-mk3s-mk3s-pi-camera-v21-mount-with-optional-) I wanted to print would solve the issue because it makes the camera's correct physical orientation. 😬

cozmo14047, 2023-06-09:
I really want this too

kanzelsberger, 2024-02-11:
This is a must and should be fairly simple to implement. I tried to rotate raspberry camera v2 using /boot/config.txt with no luck...

x23piracy, 2024-02-14:
upvote, please add this

cozmo14047, 2024-02-17:
It doesent look like this is going to be done. @TojikCZ

TojikCZ, 2024-02-17:
Yup, most likely not on PrusaLink side, sorry. Rotating a 4k photo in raw yuyv either needs a better programmer or a lot of power

kanzelsberger, 2024-02-17:
Well for those running PrusaLink on Raspberry Pi 3, 4 or 5 (probably also Zero 2) there is enough power. However, if that is complicated, why not do that on the frontend side? I would say it will be way too easy to do it in JavaScript when the image is received. It's one operation every 10 seconds in worst case, that is not very expensive.

TojikCZ, 2024-02-19:
That's what i was thinking

davidstosik, 2024-02-26:
Aren't there ways, with Raspberry Pi and Camera, to rotate the image at the driver level?

Things I found googling:
- `alias raspivid="/usr/bin/raspivid -rot 90"` ([source](https://forums.raspberrypi.com/viewtopic.php?t=273094))
- `v4l2-ctl --set-ctrl=rotate=90` ([source](https://raspberrypi.stackexchange.com/questions/111713/is-there-a-way-to-rotate-the-cameras-image-at-the-driver-level-not-processing))
- Does libcamera.Transform apply at the driver level? ([source](https://raspberrytips.com/picamera2-raspberry-pi/#:~:text=I%20want%20to%20rotate%20the%20picture%20to%20180%C2%B0%2C))

There seem to be plenty of options that are not "transpose the image data _after_ taking the capture", though I'm not knowledgeable enough to confirm what you'd need in PrusaLink... 🤔

cozmo14047, 2024-02-26:
> Aren't there ways, with Raspberry Pi and Camera, to rotate the image at the driver level?
> 
> Things I found googling:
> 
> * `alias raspivid="/usr/bin/raspivid -rot 90"` ([source](https://forums.raspberrypi.com/viewtopic.php?t=273094))
> * `v4l2-ctl --set-ctrl=rotate=90` ([source](https://raspberrypi.stackexchange.com/questions/111713/is-there-a-way-to-rotate-the-cameras-image-at-the-driver-level-not-processing))
> * Does libcamera.Transform apply at the driver level? ([source](https://raspberrytips.com/picamera2-raspberry-pi/#:~:text=I%20want%20to%20rotate%20the%20picture%20to%20180%C2%B0%2C))
>   
>   * [picamera2/examples/rotation.py](https://github.com/raspberrypi/picamera2/blob/main/examples/rotation.py): `preview_config["transform"] = libcamera.Transform(hflip=1, vflip=1)` (it also
> 
> There seem to be some options that are not "transpose the image data _after_ taking the capture", though I'm not knowledgeable enough to confirm what you'd need in PrusaLink... 🤔
> 
> Another idea: **EXIF data**. 💡 When I rotate my iPhone (or digital camera) and take a picture, the file's data itself is not rotated so that the JPEG binary at the beginning of the file is the actual top left corner of my picture. Instead, an EXIF tag is added to the JPEG file:
> 
> ![exif orientation](https://private-user-images.githubusercontent.com/816901/307634687-81fe155b-7dd2-4c18-a7a4-47777d511bf5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg5MzQzNzksIm5iZiI6MTcwODkzNDA3OSwicGF0aCI6Ii84MTY5MDEvMzA3NjM0Njg3LTgxZmUxNTViLTdkZDItNGMxOC1hN2E0LTQ3Nzc3ZDUxMWJmNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyNlQwNzU0MzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mN2ZiOGRiZTQ0NzgyZmVmOGZlZjExZDJjZjA4OGM3MGVlZjQzMzNjMmE5NjJhZGYwMGJkN2E1ZmU4ZWFjNzg2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.KO3ujGz5VIS2L4cWiplTcL44eTsMnc5RCCPMb-b0ey0) ([source](https://jdhao.github.io/2019/07/31/image_rotation_exif_info/#exif-orientation-flag))
> 
> `camera.exif:tags` should allow the addition of the `Orientation` tag ([doc](https://picamera.readthedocs.io/en/release-1.13/api_camera.html#:~:text=Make%2C%20Model%2C%20StripOffsets%2C-,Orientation,-%2C%20SamplesPerPixel%2C%20RowsPerString%2C%20StripByteCounts))
> 
> I don't know how to edit the Python code directly on my Raspberry Pi Zero (I can SSH and found prusa-link, but it all seems to be compiled into .pyc files and my knowledge around that is basically none). Would something like this impact the output picture? (Having installed the exif package with eg. `pip install exif`.)
> 
> ```diff
> --- prusa/link/cameras/encoders.py  2024-02-26 11:44:40.581695756 +0900
> +++ prusa/link/cameras/encoders.py 2024-02-26 11:34:40.280423311 +0900
> @@ -15,6 +15,8 @@
>  import numpy as np
>  from turbojpeg import TJSAMP_422, TurboJPEG  # type: ignore
> 
> +from exif import Image as exifImage
> +
>  from . import v4l2
> 
>  jpeg = TurboJPEG()
> @@ -357,6 +359,12 @@
>              raise RuntimeError(
>                  "Encoding failed - re-queueing the coded buffer")
> 
> +        image = exifImage(output)
> +        image.set("Orientation", 3)
> +        output = image.get_file()
> +
>          return output
> ```
> 
> I'd expect the image displayed in the PrusaLink web interface to be upside down.
> 
> Update: it works! Basically, removing `encoders.pyc` and copying `encoders.py` from the repository allowed me to run PrusaLink with the modified file (need to restart prusalink with `prusalink restart`). Of course, this is just a proof of concept and it would need more work (add a setting somewhere, store it, and use that setting instead of the hard-coded `3`), but as a proof of concept, it shows that my Raspberry Pi Zero W was able to mark the image as rotated 180º using an EXIF tag (without having to transpose the whole binary data).
> 
> Here's the kind of image it showed in the UI:
> 
> <img alt="image" width="317" src="https://private-user-images.githubusercontent.com/816901/307653191-9e7b720e-b8db-4e8b-8a31-554ce97a9916.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg5MzQzNzksIm5iZiI6MTcwODkzNDA3OSwicGF0aCI6Ii84MTY5MDEvMzA3NjUzMTkxLTllN2I3MjBlLWI4ZGItNGU4Yi04YTMxLTU1NGNlOTdhOTkxNi5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjZUMDc1NDM5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTE1N2JmMjRkMmJiNWI4NTNkYWI3ZTAyMTczNTk0YzQwMjBlNWU2YTE5NGYzY2MxM2IyZWQ4OTJiN2YyM2RiZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.B_2YADZhsuHkLjnQ8qfyHc3XXqv1wpUbHWkMrPvzfxg">
> Whereas without my hack, the image looks like this:
> 
> <img alt="image" width="317" src="https://private-user-images.githubusercontent.com/816901/307653554-957e21e4-3e32-49fe-a4f5-05e9bc7d7f45.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg5MzQzNzksIm5iZiI6MTcwODkzNDA3OSwicGF0aCI6Ii84MTY5MDEvMzA3NjUzNTU0LTk1N2UyMWU0LTNlMzItNDlmZS1hNGY1LTA1ZTliYzdkN2Y0NS5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16L [... cut ...] 1BRSzRaQSUyRjIwMjQwMjI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyNlQwNzU0MzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mNGEwMzNjZjk4YmE1YzFlZDYyZGY5NDQxNWZlODUwZDUzMzRhOTdkNjNiN2E3YTczYWY0Y2RhMmUzZDQwMzljJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.3vGYPaZxQe7DtHV7L_sUrFVYOI4IZwohzTVmZwYHC6Y">

Good job, you were actully able to do it ( somthing the devs couldnt do) @TojikCZ . I will be looking at doing this, maybe add a pull request