prusa3d/Prusa-Link #758: Updating through the UI does not appear to currently work

darraghbr, 2023-05-19:
When I go to the local IP address of the printer and go Settings / System Updates / Check Updates 

I currently get the following message:

[image]

Is this expected behaviour? The image in the release notes suggests that this should be working? 

Do I need to update manually?

Or is it just because the update has not trickled down to the appropriate area yet?

timemaster5, 2023-05-22:
Can we have some instructions on how to upgrade please?

TojikCZ, 2023-05-22:
Hi :/ a full image will be needed, the updater in RC3 is borked and who knows hot the 0.7.0 itself will hold up. We do not do full system updates, that is exceptionally hard, so we are hoping to get a few PrusaLink updates before requiring an image re-flash anyways.

darraghbr, 2023-05-27:
Is there any way to only flash the required image onto the card and leave any uploaded G-code files? That's the annoying bit for me anyway, using the config.ini files does 90% of the work but I've lost every G-code that I've uploaded every time I have to update Prusa Link.

TojikCZ, 2023-12-01:
This is fixed as far as i know. It cannot always work, but minor updates and bug fixes can now be pushed through OTA and installed from the web interface