prusa3d/Prusa-Link #769: Use HTTPS instead of HTTP

shravan2x, 2023-05-29:
Could an option be added to run the PrusaLink server over HTTPS? It would make HTTP authentication more secure

ondratu, 2023-05-30:
That would be really great, but problem with local TLS certificates is, that there will be self-signed, or very difficult to manage. So at this moment, we use HTTP Digest, which is easiest way to implement on all platforms we use, with basic security options.

Another (best) way is stop using local web interface - which can be disable in future, and use only Connect. All our printers communicate with Connect over TLS or over encrypted way.

closed 2023-05-30 by ondratu

shravan2x, 2023-05-30:
My concern with using Connect is create a dependence on a cloud server to use my own product. This creates concerns like:

1. Privacy of models being printed
2. Protection against Prusa cloud servers being shut down in the future
3. Protection against internet provider downtime

Even using a self-signed certificate is good as the connection data is encrypted. You could support both the HTTP and HTTPS ports, which is what HP OfficeJet printers do

shravan2x, 2023-07-07:
@ondratu could you please consider my response above?