prusa3d/Prusa-Link #773: Kernel oops when using picam

jbebel, 2023-06-02:
This started in 0.7.0rc3, but I didn't examine it until 0.7.0. With a picam v2 attached to a pi zero W, prusalink starts up, but soon hangs and no longer responds on the web interface. I finally decided to go against the installation guide and kill a kitten by enabling ssh (not sure why ssh is so strongly discouraged), and I find that when it stop responding, there is a kernel oops in the syslog:

`Jun  2 09:06:04 prusa kernel: [  327.801151] 8<--- cut here ---
Jun  2 09:06:04 prusa kernel: [  327.801237] Unable to handle kernel paging request at virtual address b6d955c8
Jun  2 09:06:04 prusa kernel: [  327.801275] [b6d955c8] *pgd=02ef0831, *pte=0248618f, *ppte=02486a2e
Jun  2 09:06:04 prusa kernel: [  327.801329] Internal error: Oops: 81f [#2] ARM
Jun  2 09:06:04 prusa kernel: [  327.801357] Modules linked in: imx219 brcmfmac vc4 nft_chain_nat xt_REDIRECT nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 brcmutil xt_tcpudp nft_compat nf_tables 8021q garp stp llc sha256_generic snd_soc_hdmi_codec libsha256 drm_display_helper nfnetlink cec cfg80211 drm_dma_helper drm_kms_helper snd_soc_core bcm2835_unicam i2c_mux_pinctrl i2c_mux v4l2_dv_timings bcm2835_codec(C) snd_compress v4l2_fwnode snd_pcm_dmaengine syscopyarea raspberrypi_hwmon v4l2_async sysfillrect bcm2835_v4l2(C) sysimgblt fb_sys_fops v4l2_mem2mem bcm2835_isp(C) bcm2835_mmal_vchiq(C) videobuf2_vmalloc rfkill videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common snd_bcm2835(C) videodev snd_pcm snd_timer i2c_bcm2835 snd mc vc_sm_cma(C) uio_pdrv_genirq fixed uio drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
Jun  2 09:06:04 prusa kernel: [  327.801815] CPU: 0 PID: 716 Comm: pl#Photographer Tainted: G      D  C         6.1.21+ #1642
Jun  2 09:06:04 prusa kernel: [  327.801852] Hardware name: BCM2835
Jun  2 09:06:04 prusa kernel: [  327.801872] PC is at vector_swi+0x5c/0x64
Jun  2 09:06:04 prusa kernel: [  327.801912] LR is at vector_swi+0x50/0x64
Jun  2 09:06:04 prusa kernel: [  327.801938] pc : [<c000837c>]    lr : [<c0008370>]    psr: 60000013
Jun  2 09:06:04 prusa kernel: [  327.801962] sp : dca49fb0  ip : 00c5387d  fp : 00000000
Jun  2 09:06:04 prusa kernel: [  327.801983] r10: 00c5387d  r9 : b6d95588  r8 : c0008408
Jun  2 09:06:04 prusa kernel: [  327.802005] r7 : 00000193  r6 : b6fb0880  r5 : 00000001  r4 : a1b72598
Jun  2 09:06:04 prusa kernel: [  327.802029] r3 : 00000000  r2 : b6e49000  r1 : a1b72598  r0 : 00000001
Jun  2 09:06:04 prusa kernel: [  327.802053] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Jun  2 09:06:04 prusa kernel: [  327.802081] Control: 00c5387d  Table: 06ad0008  DAC: 00000055
Jun  2 09:06:04 prusa kernel: [  327.802103] Register r0 information: non-paged memory
Jun  2 09:06:04 prusa kernel: [  327.802133] Register r1 information: non-paged memory
Jun  2 09:06:04 prusa kernel: [  327.802159] Register r2 information: non-paged memory
Jun  2 09:06:04 prusa kernel: [  327.802185] Register r3 information: NULL pointer
Jun  2 09:06:04 prusa kernel: [  327.802209] Register r4 information: non-paged memory
Jun  2 09:06:04 prusa kernel: [  327.802234] Register r5 information: non-paged memory
Jun  2 09:06:04 prusa kernel: [  327.802260] Register r6 information: non-paged memory
Jun  2 09:06:04 prusa kernel: [  327.802284] Register r7 information: non-paged memory
Jun  2 09:06:04 prusa kernel: [  327.802309] Register r8 information: non-slab/vmalloc memory
Jun  2 09:06:04 prusa kernel: [  327.802338] Register r9 information: non-paged memory
Jun  2 09:06:04 prusa kernel: [  327.802364] Register r10 information: non-paged memory
Jun  2 09:06:04 prusa kernel: [  327.802389] Register r11 information: NULL pointer
Jun  2 09:06:04 prusa kernel: [  327.802413] Register r12 information: non-paged memory
Jun  2 09:06:04 prusa kernel: [  327.802439] Process pl#Photographer (pid: 716, stack limit = 0xfe563b0c)
Jun  2 09:06:04 prusa kernel: [  327.802474] Stack: (0xdca49fb0 to 0xdca4a000)
Jun  2 09:06:04 prusa kernel: [  327.802504] 9fa0:                                     00000001 a1b72598 b6e49000 00000000
Jun  2 09:06:04 prusa kernel: [  327.802539] 9fc0: a1b72598 00000001 b6fb0880 00000193 a1b7261c 0000000f a1b727c8 00000000
Jun  2 09:06:04 prusa kernel: [  327.802575] 9fe0: b6e49000 a1b72568 b6d95630 b6d95588 60000010 00000001 00000000 00000000
Jun  2 09:06:04 prusa kernel: [  327.802615] Code: eb03a682 f1080080 ee1d9f70 e28f8088 (e5897040)
Jun  2 09:06:04 prusa kernel: [  327.802644] ---[ end trace 0000000000000000 ]---
`

Are there other logs that would be helpful to see? I fear that I'll have to disconnect my picam just to get prusalink to run for now.

jbebel, 2023-06-02:
The memory seems to be ok. I disconnected the picam and got everything set up and working fine, shut it down, reconnected the picam, and restarted. I got logged in enough to see that roughly half the memory was free, and no swap was in use, but after a while, I got a a couple kernel oopses back to back.

```
Jun  2 11:13:47 prusa kernel: [  531.771563] 8<--- cut here ---
Jun  2 11:13:47 prusa kernel: [  531.771646] Unable to handle kernel paging request at virtual address b6d245c8
Jun  2 11:13:47 prusa kernel: [  531.771685] [b6d245c8] *pgd=0694a831, *pte=0258d18f, *ppte=0258da2e
Jun  2 11:13:47 prusa kernel: [  531.771737] Internal error: Oops: 81f [#2] ARM
Jun  2 11:13:47 prusa kernel: [  531.771765] Modules linked in: imx219 brcmfmac vc4 brcmutil nft_chain_nat xt_REDIRECT 8021q garp nf_nat stp llc nf_conntrack nf_defrag_ipv6 sha256_generic snd_soc_hdmi_codec nf_defrag_ipv4 libsha256 drm_display_helper xt_tcpudp cec nft_compat nf_tables nfnetlink drm_dma_helper cfg80211 drm_kms_helper snd_soc_core bcm2835_unicam snd_compress snd_pcm_dmaengine v4l2_dv_timings i2c_mux_pinctrl syscopyarea i2c_mux sysfillrect raspberrypi_hwmon bcm2835_codec(C) v4l2_fwnode sysimgblt fb_sys_fops v4l2_async bcm2835_v4l2(C) v4l2_mem2mem bcm2835_isp(C) bcm2835_mmal_vchiq(C) rfkill videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common snd_bcm2835(C) snd_pcm videodev i2c_bcm2835 snd_timer snd mc vc_sm_cma(C) uio_pdrv_genirq fixed uio drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
Jun  2 11:13:47 prusa kernel: [  531.772218] CPU: 0 PID: 746 Comm: pl#Photographer Tainted: G      D  C         6.1.21+ #1642
Jun  2 11:13:47 prusa kernel: [  531.772255] Hardware name: BCM2835
Jun  2 11:13:47 prusa kernel: [  531.772274] PC is at vector_swi+0x5c/0x64
Jun  2 11:13:47 prusa kernel: [  531.772314] LR is at vector_swi+0x50/0x64
Jun  2 11:13:47 prusa kernel: [  531.772341] pc : [<c000837c>]    lr : [<c0008370>]    psr: 60000013
Jun  2 11:13:47 prusa kernel: [  531.772365] sp : dca15fb0  ip : 00c5387d  fp : 00000000
Jun  2 11:13:47 prusa kernel: [  531.772386] r10: 00c5387d  r9 : b6d24588  r8 : c0008408
Jun  2 11:13:47 prusa kernel: [  531.772408] r7 : 00000193  r6 : b6f3f880  r5 : 00000001  r4 : a3b61598
Jun  2 11:13:47 prusa kernel: [  531.772431] r3 : 00000000  r2 : b6dd8000  r1 : a3b61598  r0 : 00000001
Jun  2 11:13:47 prusa kernel: [  531.772455] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Jun  2 11:13:47 prusa kernel: [  531.772483] Control: 00c5387d  Table: 02bf8008  DAC: 00000055
Jun  2 11:13:47 prusa kernel: [  531.772504] Register r0 information: non-paged memory
Jun  2 11:13:47 prusa kernel: [  531.772533] Register r1 information: non-paged memory
Jun  2 11:13:47 prusa kernel: [  531.772559] Register r2 information: non-paged memory
Jun  2 11:13:47 prusa kernel: [  531.772584] Register r3 information: NULL pointer
Jun  2 11:13:47 prusa kernel: [  531.772608] Register r4 information: non-paged memory
Jun  2 11:13:47 prusa kernel: [  531.772632] Register r5 information: non-paged memory
Jun  2 11:13:47 prusa kernel: [  531.772657] Register r6 information: non-paged memory
Jun  2 11:13:47 prusa kernel: [  531.772682] Register r7 information: non-paged memory
Jun  2 11:13:47 prusa kernel: [  531.772707] Register r8 information: non-slab/vmalloc memory
Jun  2 11:13:47 prusa kernel: [  531.772736] Register r9 information: non-paged memory
Jun  2 11:13:47 prusa kernel: [  531.772760] Register r10 information: non-paged memory
Jun  2 11:13:47 prusa kernel: [  531.772785] Register r11 information: NULL pointer
Jun  2 11:13:47 prusa kernel: [  531.772810] Register r12 information: non-paged memory
Jun  2 11:13:47 prusa kernel: [  531.772836] Process pl#Photographer (pid: 746, stack limit = 0x34698c37)
Jun  2 11:13:47 prusa kernel: [  531.772872] Stack: (0xdca15fb0 to 0xdca16000)
Jun  2 11:13:47 prusa kernel: [  531.772901] 5fa0:                                     00000001 a3b61598 b6dd8000 00000000
Jun  2 11:13:47 prusa kernel: [  531.772938] 5fc0: a3b61598 00000001 b6f3f880 00000193 a3b6161c 00000010 a3b617c8 00000000
Jun  2 11:13:47 prusa kernel: [  531.772973] 5fe0: b6dd8000 a3b61568 b6d24630 b6d24588 60000010 00000001 00000000 00000000
Jun  2 11:13:47 prusa kernel: [  531.773014] Code: eb03a682 f1080080 ee1d9f70 e28f8088 (e5897040)
Jun  2 11:13:47 prusa kernel: [  531.773043] ---[ end trace 0000000000000000 ]---
Jun  2 11:13:47 prusa kernel: [  531.833784] 8<--- cut here ---
Jun  2 11:13:47 prusa kernel: [  531.833849] Unable to handle kernel paging request at virtual address bfa58b9c
Jun  2 11:13:47 prusa kernel: [  531.833885] [bfa58b9c] *pgd=00000000
Jun  2 11:13:47 prusa kernel: [  531.833925] Internal error: Oops: 80000005 [#3] ARM
Jun  2 11:13:47 prusa kernel: [  531.833955] Modules linked in: imx219 brcmfmac vc4 brcmutil nft_chain_nat xt_REDIRECT 8021q garp nf_nat stp llc nf_conntrack nf_defrag_ipv6 sha256_generic snd_soc_hdmi_codec nf_defrag_ipv4 libsha256 drm_display_helper xt_tcpudp cec nft_compat nf_tables nfnetlink drm_dma_helper cfg80211 drm_kms_helper snd_soc_core bcm2835_unicam snd_compress snd_pcm_dmaengine v4l2_dv_timings i2c_mux_pinctrl syscopyarea i2c_mux sysfillrect raspberrypi_hwmon bcm2835_codec(C) v4l2_fwnode sysimgblt fb_sys_fops v4l2_async bcm2835_v4l2(C) v4l2_mem2mem bcm2835_isp(C) bcm2835_mmal_vchiq(C) rfkill videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common snd_bcm2835(C) snd_pcm video [... cut ...]
Jun  2 11:13:47 prusa kernel: [  531.836959] 5fa0:                   00000000 b1b0c040 0000001a c044560f ae0f83bc b44f0788
Jun  2 11:13:47 prusa kernel: [  531.836995] 5fc0: 00000000 b1b0c040 ae0f8400 00000036 b11268c8 ae0f850c ae0f83bc b1b64d08
Jun  2 11:13:47 prusa kernel: [  531.837027] 5fe0: b465ed44 ae0f8384 b45dc940 b6d5bd4c
Jun  2 11:13:47 prusa kernel: [  531.837059] Code: bad PC value
Jun  2 11:13:47 prusa kernel: [  531.837086] ---[ end trace 0000000000000000 ]---
Jun  2 11:13:47 prusa kernel: [  531.837108] note: pl#main[566] exited with irqs disabled
```

jbebel, 2023-06-02:
Well, maybe it's more than the camera. I thought the camera was the issue, but I have it disconnected. All seemed stable, but I attempted to upload a gcode from prusaslicer, and I got another oops. I was uploading things successfully with rc2, so I feel like there's a regression here.

```
Jun  2 12:05:40 prusa prusa.link.web.lib.classes[489]: INFO: 10.0.4.3 - "POST /api/files/local HTTP/1.1" 401 496 {info():195}
Jun  2 12:05:42 prusa kernel: [ 2727.927071] 8<--- cut here ---
Jun  2 12:05:42 prusa kernel: [ 2727.927142] Unable to handle kernel paging request at virtual address c0e06dc8
Jun  2 12:05:42 prusa kernel: [ 2727.927178] [c0e06dc8] *pgd=00e0041e(bad)
Jun  2 12:05:42 prusa kernel: [ 2727.927218] Internal error: Oops: 8000000d [#1] ARM
Jun  2 12:05:42 prusa kernel: [ 2727.927246] Modules linked in: brcmfmac vc4 brcmutil sha256_generic libsha256 nft_chain_nat xt_REDIRECT nf_nat nf_conntrack snd_soc_hdmi_codec nf_defrag_ipv6 8021q nf_defrag_ipv4 garp stp llc xt_tcpudp drm_display_helper cfg80211 nft_compat cec nf_tables nfnetlink drm_dma_helper drm_
kms_helper snd_soc_core bcm2835_codec(C) snd_compress snd_pcm_dmaengine bcm2835_v4l2(C) syscopyarea sysfillrect bcm2835_isp(C) v4l2_mem2mem sysimgblt raspberrypi_hwmon fb_sys_fops bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops rfkill videobuf2_v4l2 videobuf2_common snd_bcm2835(C) video
dev snd_pcm i2c_bcm2835 snd_timer snd vc_sm_cma(C) mc uio_pdrv_genirq fixed uio drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
Jun  2 12:05:42 prusa kernel: [ 2727.927649] CPU: 0 PID: 59 Comm: kworker/u3:0 Tainted: G         C         6.1.21+ #1642
Jun  2 12:05:42 prusa kernel: [ 2727.927684] Hardware name: BCM2835
Jun  2 12:05:42 prusa kernel: [ 2727.927705] Workqueue: brcmf_wq/mmc1:0001:1 brcmf_sdio_dataworker [brcmfmac]
Jun  2 12:05:42 prusa kernel: [ 2727.928268] PC is at buffer+0x2cd8/0x3e80
Jun  2 12:05:42 prusa kernel: [ 2727.928308] LR is at brcmf_sdio_dataworker+0x924/0x2468 [brcmfmac]
Jun  2 12:05:42 prusa kernel: [ 2727.928732] pc : [<c0e06dc8>]    lr : [<bf968880>]    psr: 60000113
Jun  2 12:05:42 prusa kernel: [ 2727.928758] sp : dc965e18  ip : 00000000  fp : c4d2ee10
Jun  2 12:05:42 prusa kernel: [ 2727.928781] r10: c4d2ed54  r9 : 00000075  r8 : bf97ca04
Jun  2 12:05:42 prusa kernel: [ 2727.928802] r7 : 00000000  r6 : 00000000  r5 : c100f800  r4 : 00000002
Jun  2 12:05:42 prusa kernel: [ 2727.928826] r3 : dd061882  r2 : 00000000  r1 : 00000000  r0 : c100f800
Jun  2 12:05:42 prusa kernel: [ 2727.928849] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Jun  2 12:05:42 prusa kernel: [ 2727.928877] Control: 00c5387d  Table: 043c8008  DAC: 00000055
Jun  2 12:05:42 prusa kernel: [ 2727.928897] Register r0 information: slab kmalloc-2k start c100f800 pointer offset 0 size 2048
Jun  2 12:05:42 prusa kernel: [ 2727.928958] Register r1 information: NULL pointer
Jun  2 12:05:42 prusa kernel: [ 2727.928986] Register r2 information: NULL pointer
Jun  2 12:05:42 prusa kernel: [ 2727.929010] Register r3 information: non-paged memory
Jun  2 12:05:42 prusa kernel: [ 2727.929036] Register r4 information: non-paged memory
Jun  2 12:05:42 prusa kernel: [ 2727.929061] Register r5 information: slab kmalloc-2k start c100f800 pointer offset 0 size 2048
Jun  2 12:05:42 prusa kernel: [ 2727.929110] Register r6 information: NULL pointer
Jun  2 12:05:42 prusa kernel: [ 2727.929135] Register r7 information: NULL pointer
Jun  2 12:05:42 prusa kernel: [ 2727.929159] Register r8 information: 74-page vmalloc region starting at 0xbf945000 allocated at load_module+0x888/0x1d04
Jun  2 12:05:42 prusa kernel: [ 2727.929214] Register r9 information: non-paged memory
Jun  2 12:05:42 prusa kernel: [ 2727.929241] Register r10 information: slab kmalloc-1k start c4d2ec00 pointer offset 340 size 1024
Jun  2 12:05:42 prusa kernel: [ 2727.929293] Register r11 information: slab kmalloc-1k start c4d2ec00 pointer offset 528 size 1024
Jun  2 12:05:42 prusa kernel: [ 2727.929345] Register r12 information: NULL pointer
Jun  2 12:05:42 prusa kernel: [ 2727.929371] Process kworker/u3:0 (pid: 59, stack limit = 0x760c500d)
Jun  2 12:05:42 prusa kernel: [ 2727.929405] Stack: (0xdc965e18 to 0xdc966000)
Jun  2 12:05:42 prusa kernel: [ 2727.929431] 5e00:                                                       17efa04e 0000027b
Jun  2 12:05:42 prusa kernel: [ 2727.929466] 5e20: dc965ec0 c100f800 c11e5010 dd061882 00000001 00000002 c282ee40 00000000
Jun  2 12:05:42 prusa kernel: [ 2727.929501] 5e40: dc965ec0 bf97ca04 00000075 c4d2ed54 c4d2ee10 bf968880 00000005 c069d0dc
Jun  2 12:05:42 prusa kernel: [ 2727.929537] 5e60: c0c3b438 c0053a74 c0c3b438 c18eb268 00000000 c4d2ec00 00000031 c4d2ed54
Jun  2 12:05:42 prusa kernel: [ 2727.929571] 5e80: c4d2ee10 00000075 c4d2ec10 00000000 00000014 00000040 00000000 00000032
Jun  2 12:05:42 prusa kernel: [ 2727.929605] 5ea0: c4d2ed36 c0c3b420 00000000 c0053838 00000000 c18eb0c0 c1935140 00000000
Jun  2 12:05:42 prusa kernel: [ 2727.929639] 5ec0: 06000277 0000bf18 c0c3b40e c0043c00 00000000 ffffffff 00000002 c18eb0c0
Jun  2 12:05:42 prusa kernel: [ 2727.929674] 5ee0: c0c3b420 c1930000 c2996380 00000000 dc965f4c c08c6378 c003c43c 40000113
Jun  2 12:05:42 prusa kernel: [ 2727.929709] 5f00: ffffffff dc965f44 25092bd8 dd061882 00000000 c4d2ee10 c18285a0 00000060
Jun  2 12:05:42 prusa kernel: [ 2727.929743] 5f20: c4dab100 00000000 c4dab105 c4d2ee14 c1228a00 c003c10c c1228a18 c0ca14c0
[... cut ...]
Jun  2 12:05:42 prusa kernel: [ 2727.929948] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
Jun  2 12:05:42 prusa kernel: [ 2727.930017]  brcmf_sdio_dataworker [brcmfmac] from __schedule+0x308/0x95c
Jun  2 12:05:42 prusa kernel: [ 2727.930536]  __schedule from 0xc1228a18
Jun  2 12:05:42 prusa kernel: [ 2727.930607] Code: 00000000 00000000 00000000 00000000 (00000000)
Jun  2 12:05:42 prusa kernel: [ 2727.930639] ---[ end trace 0000000000000000 ]---
Jun  2 12:05:42 prusa kernel: [ 2727.930661] note: kworker/u3:0[59] exited with irqs disabled
```

TojikCZ, 2023-06-02:
Welp, this looks like a hardware fault to me. And apparently to ChatGPT as well. I tried to install the image anyway and no kernel panic for me.

[image]

Could you check for HW faults? Connect a different power supply, try a different card or try a different board? Would be nice to try and isolate it. If you feel like it, get something else on the pi, smack it with a load and see whether the errors stop.

jbebel, 2023-06-02:
I see. That's a bummer. I have a single Pi 0 W, and a single Prusa i3 Mk3S+, and it's connected via the onboard headers, so the Prusa provides the power. My only point of comparison is that I never saw this happening with rc2, so maybe it was coincidental that it started happening with the upgrade to rc3. I'll try going back to rc2 and see if I can reproduce it under load. If so, I'll blame the hardware. Too bad it's so impossible to obtain a new raspberry pi now.

TojikCZ, 2023-06-02:
Also try to connect the pi using usb, maybe the power provided by your printer is throwing it out of whack. If you got a second SD card, try that. Change one thing and retry

TojikCZ, 2023-06-06:
How's it going? Any new info?

jbebel, 2023-06-16:
I've been using 0.7rc2 for a while with no crashes yet, everything else the same, using the Mk3s+ as the power source still. It could still be a hardware problem, but there's something in rc3 and 0.7 that brings it out..