prusa3d/Prusa-Link #774: Connecting webcam causes system to crash

neoyarus, 2023-06-05:
I am using a radxa zero SBC with 2GB of ram and a 4-core CPU connected to a prusa MK3S+ via the RPI0 port on the mainboard. I have a 1080p usb-c webcam, which the radxa zero should have more than enough power to drive. When i connect the webcam without prusalink running, everything is fine. However, when i connect the webcam with prusalink running, the cpu load goes up untill the device becomes unresponsive and crashes. Here are the logs from the moment when i plug in the webcam:

`rock@radxa-zero:~$ journalctl -f

-- Logs begin at Mon 2023-05-08 19:57:01 UTC. --
Jun 05 10:07:49 radxa-zero sshd[1878]: pam_env(sshd:session): Unable to open env                                                                                              file: /etc/default/locale: No such file or directory
Jun 05 10:07:49 radxa-zero systemd[1956]: Started D-Bus User Message Bus.
Jun 05 10:07:49 radxa-zero systemd[1956]: Started Sound Service.
Jun 05 10:07:49 radxa-zero systemd[1956]: Reached target Main User Target.
Jun 05 10:07:49 radxa-zero systemd[1956]: Startup finished in 1.052s.
Jun 05 10:07:49 radxa-zero bluetoothd[1714]: Endpoint registered: sender=:1.30 p                                                                                             ath=/MediaEndpoint/A2DPSink/sbc
Jun 05 10:07:49 radxa-zero bluetoothd[1714]: Endpoint registered: sender=:1.30 p                                                                                             ath=/MediaEndpoint/A2DPSource/sbc
Jun 05 10:07:49 radxa-zero kernel: Bluetooth: RFCOMM TTY layer initialized
Jun 05 10:07:49 radxa-zero kernel: Bluetooth: RFCOMM socket layer initialized
Jun 05 10:07:49 radxa-zero kernel: Bluetooth: RFCOMM ver 1.11
Jun 05 10:09:07 radxa-zero kernel: usb 1-1: new high-speed USB device number 2 using xhci-hcd
Jun 05 10:09:08 radxa-zero kernel: usb 1-1: New USB device found, idVendor=1bcf, idProduct=0001, bcdDevice=10.02
Jun 05 10:09:08 radxa-zero kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jun 05 10:09:08 radxa-zero kernel: usb 1-1: Product: FHD Webcam
Jun 05 10:09:08 radxa-zero kernel: usb 1-1: Manufacturer: HCVSight
Jun 05 10:09:08 radxa-zero kernel: uvcvideo: Found UVC 1.00 device FHD Webcam (1bcf:0001)
Jun 05 10:09:08 radxa-zero kernel: input: FHD Webcam: FHD Webcam as /devices/platform/soc/ffe09000.usb/ff500000.usb/xhci-hcd.0.auto/usb1/1-1/1-1:1.0/input/input0
Jun 05 10:09:08 radxa-zero kernel: usbcore: registered new interface driver uvcvideo
Jun 05 10:09:08 radxa-zero kernel: USB Video Class driver (1.1.1)
Jun 05 10:09:10 radxa-zero kernel: usb 1-1: Warning! Unlikely big volume range (=2880), cval->res is probably wrong.
Jun 05 10:09:10 radxa-zero kernel: usb 1-1: [5] FU [Mic Capture Volume] ch = 1, val = 0/2880/1
Jun 05 10:09:10 radxa-zero kernel: usbcore: registered new interface driver snd-usb-audio
Jun 05 10:09:10 radxa-zero systemd[1956]: Reached target Sound Card.
Jun 05 10:09:10 radxa-zero systemd-udevd[1993]: controlC1: Process '/usr/sbin/alsactl -E HOME=/run/alsa -E XDG_RUNTIME_DIR=/run/alsa/runtime restore 1' failed with exit code 99.
Jun 05 10:09:21 radxa-zero kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jun 05 10:09:24 radxa-zero kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jun 05 10:09:24 radxa-zero kernel: ieee80211 phy0: brcmf_cfg80211_get_station: GET STA INFO failed, -110
Jun 05 10:09:24 radxa-zero camera_driver[1748]: ERROR: The driver V4L2 broke while taking a photo. Disconnecting {_photo_taker():239}
                                                Traceback (most recent call last):
                                                  File "/home/rock/.local/lib/python3.9/site-packages/prusa/connect/printer/camera_driver.py", line 237, in _photo_taker
                                                    snapshot.data = self.take_a_photo()
                                                  File "/home/rock/.local/lib/python3.9/site-packages/prusa/link/cameras/v4l2_driver.py", line 514, in take_a_photo
                                                    v4l2_source_buffer = self.device.next_frame()
                                                  File "/home/rock/.local/lib/python3.9/site-packages/prusa/link/cameras/v4l2_driver.py", line 344, in next_frame
                                                    raise TimeoutError("Getting the next frame timed out")
                                                TimeoutError: Getting the next frame timed out
Jun 05 10:09:51 radxa-zero kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jun 05 10:09:55 radxa-zero kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jun 05 10:09:55 radxa-zero kernel: ieee80211 phy0: brcmf_cfg80211_get_station: GET STA INFO failed, -110
Jun 05 10:10:03 radxa-zero connect-printer[1748]: ERROR: HTTPSConnectionPool(host='connect.prusa3d.com', port=443): Read timed out. (read timeout=10) {loop_step():742}
Jun 05 10:10:04 radxa-zero kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jun 05 10:10:15 radxa-zero kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jun 05 10:10:21 radxa-zero kernel: brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
Jun 05 10:10:21 radxa-zero kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
Jun 05 10:10:31 radxa-zero kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jun 05 10:10:35 radxa-zero kernel: ieee80211 phy0: brcmf_fill_bss_param: Failed to get bss info (-110)
Jun 05 10:10:35 radxa-zero kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jun 05 10:10:35 radxa-zero kernel: ieee80211 phy0: brcmf_cfg80211_get_station: Could not get rssi (-110)
Jun 05 10:10:32 radxa-zero connect-printer[1748]: ERROR: HTTPSConnectionPool(host='connect.prusa3d.com', port=443): Read timed out. (read timeout=10) {loop_step():742}
Jun 05 10:10:49 radxa-zero connect-printer[1748]: ERROR: HTTPSConnectionPool(host='connect.prusa3d.com', port=443): Read timed out. (read timeout=10) {loop_step():742}
Jun 05 10:10:55 radxa-zero kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jun 05 10:10:55 radxa-zero kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jun 05 10:10:55 radxa-zero kernel: ieee80211 phy0: brcmf_cfg80211_get_station: GET STA INFO failed, -110
Jun 05 10:10:54 radxa-zero prusa.link.printer_adapter.lcd_printer[1748]: WARNING: Displaying an error message No Internet access  {_check_errors():289}
Jun 05 10:10:56 radxa-zero kernel: brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
Jun 05 10:10:56 radxa-zero kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
Jun 05 10:10:58 radxa-zero kernel: brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Jun 05 10:11:10 radxa-zero connect-printer[1748]: ERROR: HTTPSConnectionPool(host='connect.prusa3d.com', port=443): Max retries exceeded with url: /p/telemetry (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0xffff6c7922b0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')) {loop_step():738}
Jun 05 10:11:12 radxa-zero kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jun 05 10:11:12 radxa-zero kernel: ieee80211 phy0: brcmf_fill_bss_param: Failed to get bss info (-110)
Jun 05 10:11:10 radxa-zero systemd-resolved[1603]: Using degraded feature set (UDP) for DNS server 82.209.240.241.
Jun 05 10:11:17 radxa-zero kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jun 05 10:11:18 radxa-zero kernel: ieee80211 phy0: brcmf_cfg80211_get_station: Could not get rssi (-110)
Jun 05 10:11:21 radxa-zero kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Jun 05 10:11:27 radxa-zero kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[... 136 lines cut ...]
Jun 05 10:15:20 radxa-zero interesting_logger[1748]: WARNING: Was[DEBUG]: Gathering new value for item flow_multiplier {_confirmed():431}
[... cut ...]
Jun 05 10:15:21 radxa-zero interesting_logger[1748]: WARNING: Was[DEBUG]: Gathering new value for item print_state {_confirmed():431}
Jun 05 10:15:21 radxa-zero interesting_logger[1748]: WARNING: Was[DEBUG]: Instruction 'M27 P' enqueued to the front {_confirmed():431}
Jun 05 10:15:21 radxa-zero interesting_logger[1748]: WARNING: Was[DEBUG]: Sent to printer: b'M27 P\n' {_confirmed():431}
Jun 05 10:15:21 radxa-zero interesting_logger[1748]: WARNING: Was[DEBUG]: Printer says: 'Not SD printing' {_confirmed():431}
Jun 05 10:15:21 radxa-zero interesting_logger[1748]: WARNING: Was[DEBUG]: Ma
`

TojikCZ, 2023-06-06:
Hi, did you install PrusaLink from the pip package, or through github using the README.md file?
I won't be allowed to buy this SBC and burn time de-bugging this, so the most I can do is try and help you here.
What OS did you install before you installed PrusaLink? Does the camera work with any other app? As in, are you sure it is PrusaLink, not a misconfigured OS that is causing this.

neoyarus, 2023-06-07:
Hi, thanks for trying to help. I installed PrusaLink through github because the `pip install PrusaLink` command does not work for me. BTW, when running the `pip install git+https://github.com/prusa3d/Prusa-Connect-SDK-Printer.git` command i had to add `@0.7.0` at the end, otherwise it installs the latest dev version, which does not work (at least not on my device). 

I installed [Ubuntu Focal Server](https://github.com/radxa-build/radxa-zero/releases/tag/20220801-0213) 20.04.6 LTS (GNU/Linux 5.10.69-12-amlogic-g98700611d064 aarch64). It's an official build and pretty much the only option that supports using the GPIO pins for connecting to the printer (i have tried other images and it only worked with this one). 

I have tried using the webcam with motion and it was able to capture an image, but did not cause the system to crash either. 
I also tried it with ffmpeg and it was able to capture something, but only when using /dev/video1 (the webcam shows up as /dev/video1 and /dev/video2 and /dev/media0. I think video1 is YUY2 and video2 is mjpeg, not sure waht media0 is.

```
rock@radxa-zero:~$ ffmpeg -f v4l2 -i /dev/video1 -frames 1 image.jpg
ffmpeg version 4.2.7-0ubuntu0.1 Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.4.0-1ubuntu1~20.04.1)
  configuration: --prefix=/usr --extra-version=0ubuntu0.1 --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --arch=arm64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
Input #0, video4linux2,v4l2, from '/dev/video1':
  Duration: N/A, start: 2401.503444, bitrate: 165888 kb/s
    Stream #0:0: Video: rawvideo (YUY2 / 0x32595559), yuyv422, 1920x1080, 165888 kb/s, 5 fps, 5 tbr, 1000k tbn, 1000k tbc
File 'image.jpg' already exists. Overwrite ? [y/N] y
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> mjpeg (native))
Press [q] to stop, [?] for help
[swscaler @ 0xaaab0de0efb0] deprecated pixel format used, make sure you did set range correctly
Output #0, image2, to 'image.jpg':
  Metadata:
    encoder         : Lavf58.29.100
    Stream #0:0: Video: mjpeg, yuvj422p(pc), 1920x1080, q=2-31, 200 kb/s, 5 fps, 5 tbn, 5 tbc
    Metadata:
      encoder         : Lavc58.54.100 mjpeg
    Side data:
      cpb: bitrate max/min/avg: 0/0/200000 buffer size: 0 vbv_delay: -1
frame=    1 fps=0.0 q=5.8 Lsize=N/A time=00:00:00.20 bitrate=N/A speed=0.864x
video:71kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
rock@radxa-zero:~$ ffmpeg -f v4l2 -i /dev/video2 -frames 1 image.jpg
ffmpeg version 4.2.7-0ubuntu0.1 Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.4.0-1ubuntu1~20.04.1)
  configuration: --prefix=/usr --extra-version=0ubuntu0.1 --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --arch=arm64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
[... cut ...]
  libavfilter     7. 57.100 /  7. 57.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
[video4linux2,v4l2 @ 0xaaaab7e2b790] ioctl(VIDIOC_QUERYCAP): Inappropriate ioctl for device
/dev/media0: Inappropriate ioctl for device
```

Not sure if the OS is 100% to blame here, but as far as i know its completely untested so it might be. Is there anything else i can do to diagnose/fix this?

neoyarus, 2023-06-07:
its 1080p with a 1m type-c cable :)

TojikCZ, 2023-06-07:
What focus did you choose. If I understand it correctly, it dictates the FOV, Is the 3.6 usable, or is it too zoomed in?

neoyarus, 2023-06-07:
I also asked the seller if it needed any specific drivers for Linux and he have me this [dropbox link](https://www.dropbox.com/s/s24pqdl2cklpflq/USB%20camera.rar?spm=a2g2w.orderlist.0.0.53234aa6c137xT&dl=0). Does not look like Linux drivers to me but that's what we've got. Btw the camera works fine on both windows and android.

TojikCZ, 2023-06-07:
says uvcvideo, the same thing linux uses, so I think that's fine

TojikCZ, 2023-06-07:
Seems I can only get it if I pay out of pocket. Will wait what your investigation results in.

TojikCZ, 2023-06-09:
Okay, that would tell me that PrusaLink might, or might not be the culprit. If you can, you could modify the run command in rc.local to include "-d" in the prusalink start command, then send me the log from that, that way i might see something fail, but i'm so far more leaning towards the sbc just bogging itself down with errors of some sort. Not sure what tho. You can also try and let it run, then get the syslog after a longer period of time has passed from the SD or once you boot up again. It should be in /var/log/syslog

neoyarus, 2023-06-09:
Ran it with the -d argument and it generated an insane amount of logs. Here they are from just before i plugged in the camera and untill it dropped me out of the SSH session.

```
Jun 09 14:26:45 radxa-zero prusa.link.serial.serial_queue[1766]: DEBUG: Instruction 'M27 P' confirmed {_confirmed():420}
Jun 09 14:26:45 radxa-zero prusa.link.printer_adapter.structures.item_updater[1766]: DEBUG: Value of item print_state has been determined to be PrintState.NOT_SD_PRINTING {set_value():312}
Jun 09 14:26:45 radxa-zero prusa.link.printer_adapter.structures.item_updater[1766]: DEBUG: Scheduling invalidation of item print_state for 1s in the future {schedule_invalidation():354}
Jun 09 14:26:45 radxa-zero prusa.link.printer_adapter.state_manager[1766]: DEBUG: Default expected state change is overridden {wrapper():81}
Jun 09 14:26:45 radxa-zero urllib3.connectionpool[1766]: DEBUG: https://connect.prusa3d.com:443 "POST /p/telemetry HTTP/1.1" 204 0 {_make_request():428}
Jun 09 14:26:45 radxa-zero connect-printer[1766]: DEBUG: Telemetry response:  {send():74}
Jun 09 14:26:45 radxa-zero connect-printer[1766]: DEBUG: Sending Telemetry: <Telemetry:: at 281472969773696> {'state': 'IDLE'} {send():67}
Jun 09 14:26:45 radxa-zero urllib3.connectionpool[1766]: DEBUG: https://connect.prusa3d.com:443 "POST /p/telemetry HTTP/1.1" 204 0 {_make_request():428}
Jun 09 14:26:45 radxa-zero connect-printer[1766]: DEBUG: Telemetry response:  {send():74}
Jun 09 14:26:45 radxa-zero connect-printer[1766]: DEBUG: Sending Telemetry: <Telemetry:: at 281473231973200> {'state': 'IDLE'} {send():67}
Jun 09 14:26:45 radxa-zero urllib3.connectionpool[1766]: DEBUG: https://connect.prusa3d.com:443 "POST /p/telemetry HTTP/1.1" 204 0 {_make_request():428}
Jun 09 14:26:45 radxa-zero connect-printer[1766]: DEBUG: Telemetry response:  {send():74}
Jun 09 14:26:46 radxa-zero connect-printer[1766]: DEBUG: Sending Telemetry: <Telemetry:: at 281472969746128> {'state': 'IDLE'} {send():67}
Jun 09 14:26:46 radxa-zero my_camera_configurator[1766]: DEBUG: Running the camera governance routine {_govern():31}
Jun 09 14:26:46 radxa-zero prusa.link.cameras.v4l2_driver[1766]: DEBUG: Camera id is FHD Webcam: FHD Webcam  {_scan():419}
Jun 09 14:26:46 radxa-zero urllib3.connectionpool[1766]: DEBUG: https://connect.prusa3d.com:443 "POST /p/telemetry HTTP/1.1" 204 0 {_make_request():428}
Jun 09 14:26:46 radxa-zero connect-printer[1766]: DEBUG: Telemetry response:  {send():74}
Jun 09 14:26:46 radxa-zero connect-printer[1766]: DEBUG: Sending Telemetry: <Telemetry:: at 281473231973200> {'state': 'IDLE'} {send():67}
Jun 09 14:26:46 radxa-zero prusa.link.cameras.v4l2_driver[1766]: DEBUG: Camera id is FHD Webcam: FHD Webcam  {_scan():419}
Jun 09 14:26:46 radxa-zero urllib3.connectionpool[1766]: DEBUG: https://connect.prusa3d.com:443 "POST /p/telemetry HTTP/1.1" 204 0 {_make_request():428}
Jun 09 14:26:46 radxa-zero connect-printer[1766]: DEBUG: Telemetry response:  {send():74}
Jun 09 14:26:46 radxa-zero connect-printer[1766]: DEBUG: Sending Telemetry: <Telemetry:: at 281472969839136> {'state': 'IDLE'} {send():67}
Jun 09 14:26:46 radxa-zero prusa.link.printer_adapter.structures.item_updater[1766]: DEBUG: Item speed_multiplier has been invalidated {invalidate():256}
Jun 09 14:26:46 radxa-zero prusa.link.printer_adapter.structures.item_updater[1766]: DEBUG: Gathering new value for item speed_multiplier {_gather():413}
Jun 09 14:26:46 radxa-zero prusa.link.serial.serial_queue[1766]: DEBUG: Instruction 'M220' enqueued  {enqueue_one():323}
Jun 09 14:26:46 radxa-zero prusa.link.printer_adapter.structures.item_updater[1766]: DEBUG: Item flow_multiplier has been invalidated {invalidate():256}
Jun 09 14:26:46 radxa-zero prusa.link.serial.serial_adapter[1766]: DEBUG: Sent to printer: b'M220\n' {write():335}
Jun 09 14:26:46 radxa-zero prusa.link.serial.serial_adapter[1766]: DEBUG: Printer says: '100%' {_read_continually():305}
Jun 09 14:26:46 radxa-zero prusa.link.serial.serial_parser[1766]: DEBUG: Matched RegexPairing for ^(?P<percent>\d{0,3})%$ with priority 1686320806.5769365 having 1 handler calling {(281473261390576, 281472969747136): <bound method MatchableInstruction.output_captured of Instruction 'M220'>} {fire():52}
Jun 09 14:26:46 radxa-zero prusa.link.serial.serial_adapter[1766]: DEBUG: Printer says: 'ok' {_read_continually():305}
Jun 09 14:26:46 radxa-zero prusa.link.serial.serial_parser[1766]: DEBUG: Matched RegexPairing for ^(ok.*)|(Done saving file\.)$ with priority inf having 2 handlers calling {(281473259571376, 281473242007296): <bound method SerialQueue._confirmation_handler of <prusa.link.serial.serial_queue.MonitoredSerialQueue object at 0xffff989a8700>>, 281473231891904: <function ThreadedSerialParser.decoupled.<locals>.inner at 0xffff98002dc0>} {fire():52}
Jun 09 14:26:46 radxa-zero prusa.link.serial.serial_queue[1766]: DEBUG: Instruction 'M220' confirmed {_confirmed():420}
Jun 09 14:26:46 radxa-zero prusa.link.printer_adapter.structures.item_updater[1766]: DEBUG: Value of item speed_multiplier has been determined to be 100 {set_value():312}
Jun 09 14:26:46 radxa-zero prusa.link.printer_adapter.structures.item_updater[1766]: DEBUG: Item print_state has been invalidated {invalidate():256}
Jun 09 14:26:46 radxa-zero prusa.link.printer_adapter.structures.item_updater[1766]: DEBUG: Scheduling invalidation of item speed_multiplier for 1s in the future {schedule_invalidation():354}
[... cut ...]
                                                                 self.raw_requestline = self.rfile.readline(MAX_REQUEST_SIZE)
                                                               File "/usr/lib/python3.9/socket.py", line 704, in readinto
                                                                 return self._sock.recv_into(b)
                                                             ConnectionResetError: [Errno 104] Connection reset by peer
Jun 09 13:40:53 radxa-zero prusa.link.web.lib.classes[1757]: ERROR: Parse request error. {handle():63}
```

TojikCZ, 2023-06-09:
`Jun 09 14:26:47 radxa-zero camera[1766]: DEBUG: A camera FHD Webcam: FHD Webcam has taken a photo. (157737 bytes) {_photo_handler():534}`
Yay, the camera actually does take and encode the JPEG, the callback succeeds. Did you try to download or upload anything with the radxa zero? Gine once spoke about the raspi wifi driver bogging down the whole thing, I immediatelly correlate it to that. My theory now is that the board tries to send the jpeg and that makes it drop the connection.

neoyarus, 2023-06-09:
On a related note, when i reboot everythink after this crash, everything is fine it first, but then the printer and the sbc will completely freze, even the reset button does not work. If I turn the printer off with the switch on the power supply and then turn it back on in a minute nothing works and every pixel on the printer display is lit. If i wait for like 10-20 mins and then turn it back on it works fine for a bit and then breaks the same way again. The only way i found to break the cycle is to delete the prusa_printer_settings.ini file, and the prusalink.pid file, then configure prusalink again. After that it seems to work fine, although i suspect that just powering the printer off has a chance to corrupt the prusa_printer_settings.ini file because this problem was happening randomly even before i got the webcam. Maybe in the future prusalink could automatically back up the prusa_printer_settings.ini file after the initial setup and then restore if it fails to launch?

neoyarus, 2023-06-09:
I don't know about the wifi driver. I mean, it did not bog down when i was installing prusalink from github, and seemed to be downloading pretty quickly

TojikCZ, 2023-06-09:
ah, right, keep mixing up an image with a custom install, sorry

neoyarus, 2023-06-09:
Could it be that the wifi driver is only bogging the system down when uploading? Then again, its only uploading a single picture, and i was able to do that manually without crashing everything

TojikCZ, 2023-06-09:
Not used to seeing these
```Jun 08 17:06:57 radxa-zero prusa.link.web.lib.classes[1757]: ERROR: Error for client 192.168.88.13 {handle_error():24}
                                                             Traceback (most recent call last):
                                                               File "/usr/lib/python3.9/socketserver.py", line 683, in process_request_thread
                                                                 self.finish_request(request, client_address)
                                                               File "/usr/lib/python3.9/socketserver.py", line 360, in finish_request
                                                                 self.RequestHandlerClass(request, client_address, self)
                                                               File "/usr/lib/python3.9/socketserver.py", line 747, in __init__
                                                                 self.handle()
                                                               File "/home/rock/.local/lib/python3.9/site-packages/prusa/link/web/lib/classes.py", line 54, in handle
                                                                 self.raw_requestline = self.rfile.readline(MAX_REQUEST_SIZE)
                                                               File "/usr/lib/python3.9/socket.py", line 704, in readinto
                                                                 return self._sock.recv_into(b)
                                                             ConnectionResetError: [Errno 104] Connection reset by peer```
But other than that, i have nothing :/

The display artifacts on the printer can be replicated without the board, just hold the reset button, then turn on the PSU. Your zero might be holding the reset pin for whatever reason.

Sorry but I don't think I can help much :/

TojikCZ, 2023-06-09:
Yea, that will help narrow it down too. Of course feel free to try with a usb cable when and if you get one. Good luck

neoyarus, 2023-06-09:
I do actually have a usb cable, it came with the printer. I tried it and the results were a bit different, though functionally the same. I powered the radxa zero via one of the usb ports, then plugged a hub into the other one, and a usb-a-to-b cable into the hub. I then tried running pruslink without the -d argument. It detected the camera, and the camera even showed up in the local web interface, but there was no image and when i tried to edit the camera settings all the dropdowns just said "dropdown" and i could not open them. Anyways, it declared that there was no internet access, then crashed. Here are the logs:

```
rock@radxa-zero:~$ journalctl -f
-- Logs begin at Mon 2023-05-08 19:57:11 UTC. --
Jun 09 16:23:59 radxa-zero prusa.link.printer_adapter.structures.item_updater[2000]:                              ERROR: Gather of printer_type has failed {_gather():430}
                                                                                                                  Traceback (most recent call last):
                                                                                                                    File "/home/rock/.local/lib/python3.9/site-packages/prusa/link/printer_adapter/stru                             ctures/item_updater.py", line 415, in _gather
                                                                                                                      value = item.gather_function()
                                                                                                                    File "/home/rock/.local/lib/python3.9/site-packages/prusa/link/printer_adapter/prin                             ter_polling.py", line 472, in _get_printer_type
                                                                                                                      match = self.do_matchable("M862.2 Q",
                                                                                                                    File "/home/rock/.local/lib/python3.9/site-packages/prusa/link/printer_adapter/prin                             ter_polling.py", line 426, in do_matchable
                                                                                                                      raise RuntimeError("Printer responded with something unexpected")
                                                                                                                  RuntimeError: Printer responded with something unexpected
Jun 09 16:23:59 radxa-zero prusa.link.serial.serial_queue[2000]: ERROR: Unexpected message confirmation. Ignoring {_confirmed():408}
Jun 09 16:23:59 radxa-zero prusa.link.serial.serial_queue[2000]: ERROR: Unexpected message confirmation. Ignoring {_confirmed():408}
Jun 09 16:24:00 radxa-zero prusa.link.printer_adapter.filesystem.sd_card[2000]: WARNING: Filename b00871ef-19a2-42d9-aa3e-9f5d4ff78f77_0.2mm_PLA_MK3S_ too long, using an                              alternative: b00871~1.gco - (b00871ef-19a2-42d9-aa3e-9f5d4ff78f77_0.2mm_PLA_MK3S_).gcode {alternative_filename():45}
Jun 09 16:24:00 radxa-zero prusa.link.printer_adapter.filesystem.sd_card[2000]: WARNING: Filename 5ce7e954-63e0-454b-a365-8756e0fc55e6_0.2mm_PLA_MK3S_ too long, using an                              alternative: 5ce7e9~1.gco - (5ce7e954-63e0-454b-a365-8756e0fc55e6_0.2mm_PLA_MK3S_).gcode {alternative_filename():45}
Jun 09 16:24:36 radxa-zero sshd[2102]: Accepted password for rock from 192.168.88.13 port 61442 ssh2
Jun 09 16:24:36 radxa-zero sshd[2102]: pam_unix(sshd:session): session opened for user rock by (uid=0)
Jun 09 16:24:36 radxa-zero systemd-logind[1629]: New session 3 of user rock.
Jun 09 16:24:36 radxa-zero systemd[1]: Started Session 3 of user rock.
Jun 09 16:24:36 radxa-zero sshd[2102]: pam_env(sshd:session): Unable to open env file: /etc/default/locale: No such file or directory
Jun 09 16:25:15 radxa-zero kernel: usb 1-1.2: new high-speed USB device number 7 usin g xhci-hcd
Jun 09 16:25:15 radxa-zero kernel: usb 1-1.2: New USB device found, idVendor=1bcf, idProduct=0001, bcdDevice=10.02
Jun 09 16:25:15 radxa-zero kernel: usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jun 09 16:25:15 radxa-zero kernel: usb 1-1.2: Product: FHD Webcam
Jun 09 16:25:15 radxa-zero kernel: usb 1-1.2: Manufacturer: HCVSight
Jun 09 16:25:18 radxa-zero kernel: usb 1-1.2: Warning! Unlikely big volume range (=2880), cval->res is probably wrong.
Jun 09 16:25:18 radxa-zero kernel: usb 1-1.2: [5] FU [Mic Capture Volume] ch = 1, val = 0/2880/1
Jun 09 16:25:18 radxa-zero systemd[1942]: Reached target Sound Card.
Jun 09 16:25:18 radxa-zero kernel: uvcvideo: Found UVC 1.00 device FHD Webcam (1bcf:0001)
Jun 09 16:25:18 radxa-zero systemd-udevd[2218]: controlC2: Process '/usr/sbin/alsactl -E HOME=/run/alsa -E XDG_RUNTIME_DIR=/run/alsa/runtime restore 2' failed with exit c                             ode 99.
Jun 09 16:25:18 radxa-zero kernel: input: FHD Webcam: FHD Webcam as /devices/platform/soc/ffe09000.usb/ff500000.usb/xhci-hcd.0.auto/usb1/1-1/1-1.2/1-1.2:1.0/input/input1
Jun 09 16:25:18 radxa-zero kernel: usbcore: registered new interface driver uvcvideo
Jun 09 16:25:18 radxa-zero kernel: USB Video Class driver (1.1.1)
Jun 09 16:25:18 radxa-zero kernel: uvcvideo: Failed to query (GET_CUR) UVC control 11 on unit 2: -110 (exp. 1).
Jun 09 16:25:18 radxa-zero kernel: usb 1-1.2: 3:3: cannot get freq at ep 0x86
[... cut ...]
Jun 09 16:30:01 radxa-zero prusa.link.printer_adapter.state_manager[2960]: DEBUG: Default expected state change is overridden {wrapper():81}
Jun 09 16:30:01 radxa-zero systemd[1]: prusalink.service: Main process exited, code=exited, status=1/FAILURE
Jun 09 16:30:01 radxa-zero systemd[1]: prusalink.service: Failed with result 'exit-code'.
Jun 09 16:30:01 radxa-zero connect-printer[2960]: DEBUG: Sending Telemetry: <Telemetry:: at 281473370344176> {'state': 'IDLE'} {send():67}
Jun 09 16:30:01 radxa-zero systemd[1]: prusalink.service: Scheduled restart job, restart counter is at 17.
```

TojikCZ, 2023-06-09:
Can I see the service? PrusaLink daemonizes itself, so putting it under systemd might introduce funkiness, it's the first time i see 
```
Jun 09 16:30:01 radxa-zero systemd[1]: prusalink.service: Main process exited, code=exited, status=1/FAILURE
Jun 09 16:30:01 radxa-zero systemd[1]: prusalink.service: Failed with result 'exit-code'.
Jun 09 16:30:01 radxa-zero connect-printer[2960]: DEBUG: Sending Telemetry: <Telemetry:: at 281473370344176> {'state': 'IDLE'} {send():67}
Jun 09 16:30:01 radxa-zero systemd[1]: prusalink.service: Scheduled restart job, restart counter is at 17.
```
this in the log

TojikCZ, 2023-06-09:
i know you run it as a systemd service, and you say it has worked pretty alright before, so that is most likely not the cause :/

neoyarus, 2023-06-09:
Sure, here it is:

```
[Unit]
Description=PrusaLink Service
After=network.target

[Service]
Type=simple
User=rock
ExecStart=/home/rock/.local/bin/prusalink start
Restart=always

[Install]
WantedBy=multi-user.target
```

I have to run it as a service, otherwise it does not start up at boot. It's the same on my x86 pc, not just the radxa zero.
When using the gpio pins i also add `--serial-port /dev/ttyAML1`

TojikCZ, 2023-06-09:
First time I see AML 😅

TojikCZ, 2023-06-09:
Yea, so, you don't want to do that. You are possibly constantly re-loading all the libraries of prusalink
[image]

few things to try:
as ChatGPT suggested - use `Type=forking`
You might also want to point systemd towards the pid file created in the user home directory

Alternatively - use the `-f` option - this way PrusaLink won't daemonize, instead, systemd will be left to do it - do not combine that with Type=forking

Or - use rc.local if it's there. That's the way our images work. I personally do not like it, but my boss is oldschool and is still refusing to use systemd for this.

I had a whole install script with a systemd service in here. Big sad

TojikCZ, 2023-06-09:
to be more precise, it starts a new interpreter, imports all the libraries, then attempts to spin up prusalink, but it's already running, so nothing gets done, repeat

neoyarus, 2023-06-09:
with the `type=forking` it seems that it still spawns a whole bunch of processes, just as many as with `type=simple`. But with `type=simple` it seems like new ones are opening every few seconds.

here's how it looks with `type=forking`

```
rock@radxa-zero:~$ htop

  1  [||||||||||||||||||||||                                                 27.0%]   Tasks: 30, 38 thr; 1 running
  2  [|||||||                                                                 8.2%]   Load average: 1.45 0.48 0.17
  3  [||||||                                                                  6.4%]   Uptime: 00:01:28
  4  [|||||||                                                                 8.1%]
  Mem[||||||||||||||                                                    153M/1.94G]
  Swp[                                                                       0K/0K]

ER      PRI  NI  VIRT   RES   SHR S CPU% MEM%   TIME+  Command
ck       20   0 1923M 64304 16680 S 16.7  3.2  0:15.38 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ot       19  -1 83468 41272 40248 S  5.2  2.0  0:03.82 /lib/systemd/systemd-journald
ck       20   0 1923M 64304 16680 S  7.3  3.2  0:04.03 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ck       20   0 1923M 64304 16680 S  2.1  3.2  0:01.97 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ck       20   0  4544  3336  2072 R  1.3  0.2  0:01.02 htop
ck       20   0 1923M 64304 16680 S  1.7  3.2  0:01.16 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ck       20   0 1923M 64304 16680 S  0.4  3.2  0:00.36 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ck       20   0 1923M 64304 16680 S  0.4  3.2  0:00.46 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ck       20   0 1923M 64304 16680 S  0.4  3.2  0:00.22 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ck       20   0 1923M 64304 16680 S  0.4  3.2  0:00.16 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ck       20   0 1923M 64304 16680 S  0.0  3.2  0:00.11 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ck       20   0 1923M 64304 16680 S  0.9  3.2  0:00.26 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ck       20   0 1923M 64304 16680 S  0.0  3.2  0:00.08 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ck       20   0 1923M 64304 16680 S  0.0  3.2  0:00.05 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ck       20   0 1923M 64304 16680 S  0.4  3.2  0:00.16 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ck       20   0 1923M 64304 16680 S  0.0  3.2  0:00.09 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ck       20   0 1923M 64304 16680 S  0.4  3.2  0:00.11 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ck       20   0 1923M 64304 16680 S  0.9  3.2  0:00.09 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ot       20   0  325M 17652 14816 S  0.0  0.9  0:00.61 /usr/sbin/NetworkManager --no-daemon
ck       20   0 1923M 64304 16680 S  0.0  3.2  0:00.17 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ck       20   0 1923M 64304 16680 S  0.0  3.2  0:00.22 /usr/bin/python3.9 /home/rock/.local/bin/prusalink start -d
ot       20   0  163M  9536  7020 S  0.0  0.5  0:02.96 /sbin/init
ot       20   0 21140  5288  3448 S  0.0  0.3  0:01.71 /lib/systemd/systemd-udevd
stemd-r  20   0 24652 11032  6988 S  0.0  0.5  0:00.28 /lib/systemd/systemd-resolved
ahi      20   0  6080  2592  2312 S  0.0  0.1  0:00.16 avahi-daemon: running [radxa-zero.local]
ssagebu  20   0  7192  3620  3088 S  0.0  0.2  0:00.49 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
ot       20   0  325M 17652 14816 S  0.0  0.9  0:00.00 /usr/sbin/NetworkManager --no-daemon
ot       20   0  325M 17652 14816 S  0.0  0.9  0:00.12 /usr/sbin/NetworkManager --no-daemon
ot       20   0 16944  6632  5756 S  0.0  0.3  0:00.19 /lib/systemd/systemd-logind
ot       20   0  382M 12580  8800 S  0.0  0.6  0:00.00 /usr/lib/udisks2/udisksd
ot       20   0  382M 12580  8800 S  0.0  0.6  0:00.00 /usr/lib/udisks2/udisksd
ot       20   0  382M 12580  8800 S  0.0  0.6  0:00.00 /usr/lib/udisks2/udisksd
ot       20   0  382M 12580  8800 S  0.0  0.6  0:00.00 /usr/lib/udisks2/udisksd
ot       20   0  382M 12580  8800 S  0.0  0.6  0:00.19 /usr/lib/udisks2/udisksd
ahi      20   0  5912   280     0 S  0.0  0.0  0:00.00 avahi-daemon: chroot helper
ot       20   0 12644  7712  6876 S  0.0  0.4  0:00.07 /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant
ot       20   0  2456  1632  1268 S  0.0  0.1  0:00.03 /usr/sbin/dhcpcd
ot       20   0  8116  4396  3996 S  0.0  0.2  0:00.05 /usr/lib/bluetooth/bluetoothd
ot       20   0  227M  6192  5512 S  0.0  0.3  0:00.00 /usr/lib/policykit-1/polkitd --no-debug
ot       20   0  227M  6192  5512 S  0.0  0.3  0:00.04 /usr/lib/policykit-1/polkitd --no-debug
ot       20   0  227M  6192  5512 S  0.0  0.3  0:00.09 /usr/lib/policykit-1/polkitd --no-debug
F1Help  F2Setup F3SearchF4FilterF5Tree  F6SortByF7Nice -F8Nice +F9Kill  F10Quit

```

and this is will `type=simple`

```
rock@radxa-zero:~$ htop

  1  [|||||||||||||||||||||                                                  26.0%]   Tasks: 30, 38 thr; 1 running
  2  [||||                                                                    4.6%]   Load average: 1.45 0.99 0.45
  3  [|||||                                                                   5.9%]   Uptime: 00:04:50
[... cut ...] upled.<locals>.inner at 0xffff68f061f0>}
Jun 09 18:39:38 radxa-zero prusalink[3144]: 2023-06-09 18:39:32,902 DEBUG {serial_adapter._read_continually():305} [serial_read_thread]: Printer says: 'X:0.00 Y:0.00 Z:0.15 E:0.00 Count X: 0.00 Y:0.00 Z:0.15 E:0.00'
Jun 09 18:39:38 radxa-zero prusalink[3144]: 2023-06-09 18:39:32,968 DEBUG {serial_parser.fire():52} [serial_read_thread]: Matched RegexPairing for ^X:(?P<x>-?\d+\.\d+) Y:(?P<y>-?\d+\.\d+) Z:(?P<z>-?\d+\.\d+) E:(?P<e>-?\d+\.\d+) Count X: (?P<co

neoyarus, 2023-06-09:
Trying with rc.local currently. My rc.local file looks like this

```
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

if [ -e /etc/first_boot ]; then
    /bin/sh /etc/first_boot
    rm /etc/first_boot
    reboot
fi

/home/rock/.local/bin/prusalink start -d

exit 0

```

After reboot it seems that this option still does not work

```
rock@radxa-zero:~$ prusalink status
Service not running

```

TojikCZ, 2023-06-09:
If you have the ethernet adapter and you can see it working using something like `ip address` you can try to block your wifi by running `rfkill block wifi` it still spews a bunch of errors as far as i can see

also, there are no logs indicating the service is restarting, so that looks better

Another thing to try would be to make the service type simple, but throw out the restart on exit. If you don't specify the -f option, it should attempt to restart on exception on its own. If the whole thing doesn't crash that is.

This is probably unimportant, but 3.9 is kinda old. Do a system upgrade if feeling like it. Or get octoprint
I don't want to burden you :)

TojikCZ, 2023-06-09:
sorry, have not seen the last post yet, reading now

TojikCZ, 2023-06-09:
oh right, forgot to mention, we run it under a regular user

our startup script looks like so

```# Forward the port 80 to 8080 even on the loopback, so we van ping ourselves
iptables -t nat -A PREROUTING -i wlan0 -p tcp --dport 80 -j REDIRECT --to-port 8080
iptables -t nat -A PREROUTING -i eth0 -p tcp --dport 80 -j REDIRECT --to-port 8080
iptables -t nat -I OUTPUT -p tcp -o lo -d localhost --dport 80 -j REDIRECT --to-ports 8080

set_up_port () {
   # Sets the baudrate and cancels the hangup at the end of a connection
   stty -F "$1" 115200 -hupcl;
}

message() {
   printf "M117 $2\n" > "$1"
}

username=$(id -nu 1000)
user_site=$(su $username -c "python -m site --user-site")

set_up_port "/dev/ttyAMA0"
message "/dev/ttyAMA0" "Starting PrusaLink";

$user_site/prusa/link/data/config_copy.sh
rm -f /home/$username/prusalink.pid
export PYTHONOPTIMIZE=2
su $username -c "/home/$username/.local/bin/prusalink -i start"
```

Feel free to pick what parts you like

TojikCZ, 2023-06-09:
this gets a user with uid 1000 - usually the first regular user

neoyarus, 2023-06-09:
The thing about octoprint is that for some reason it does not support pausing the print on a prusa. It just completely skips it and goes on. A filament change command works, but that's just annoying. Plus I really like the queue system and the centralized web interface that lets me control my 2 printers from anywhere.

Also i don't know that the deal is with python on my system. When i type `sudo apt install python` it installs version 3.8.something, so i have to use `sudo apt install python 3.9`, and the same goes for pip. `sudo apt update/upgrade` does not help.

Going to kill wifi and try with your startup script now

neoyarus, 2023-06-09:
still getting `service not running` after copying your script into rc.local. Going to just launch it manually and test how the webcam works

TojikCZ, 2023-06-09:
Debian (Ubuntu base OS) has a system of OS versions, so if you got a version X - you get packages for that version, limiting the recentness of them, they sometimes get backports from newer ones - the 3.9 you installed. You would need to do an OS release upgrade. `sudo do-release-upgrade` that should get you to the newest OS, but also might knock stuff out of whack on an SBC. The arm and risc SBCs are in my experience always a bit funky with big updates like this.

TojikCZ, 2023-06-09:
yeah fair

neoyarus, 2023-06-09:
idk, looks like that startup script did run, because i can now access prusalink on port 80 instead of 8080 (idk if that command needs to be run on reboot or if its enough that i ran it once during setup). Anyways, maybe i just have one user (rock), and that's why it didn't open prusalink?

I might attempt an OS update after i try everything else today. If it wrecks everything i will go to sleep and do a fresh install tomorrow.

Anyways, after killing the wifi and launching manually, i got the webcam in the web interface, but no image

logs:

```
login as: rock
rock@192.168.88.31's password:
Welcome to Ubuntu 20.04.6 LTS (GNU/Linux 5.10.69-12-amlogic-g98700611d064 aarch64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage
Last login: Fri Jun  9 19:11:23 2023 from 192.168.88.13
rock@radxa-zero:~$ prusalink status
Service not running
rock@radxa-zero:~$ prusalink start -d
journarock@radxa-zero:~$ prusalink status
Service running with pid 2129
rock@radxa-zero:~$ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.88.31  netmask 255.255.255.0  broadcast 192.168.88.255
        inet6 fe80::f4ed:4c3a:9b2:fcac  prefixlen 64  scopeid 0x20<link>
        ether 34:29:8f:98:52:8b  txqueuelen 1000  (Ethernet)
        RX packets 5040  bytes 456104 (456.1 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2361  bytes 491957 (491.9 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 107  bytes 9507 (9.5 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 107  bytes 9507 (9.5 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.88.15  netmask 255.255.255.0  broadcast 192.168.88.255
        inet6 fe80::209a:7d56:a7f:23aa  prefixlen 64  scopeid 0x20<link>
        ether b8:13:32:94:1c:3c  txqueuelen 1000  (Ethernet)
        RX packets 120  bytes 14020 (14.0 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 76  bytes 11543 (11.5 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

rock@radxa-zero:~$ rfkill unblock wifi
rfkill: cannot open /dev/rfkill: Permission denied
rock@radxa-zero:~$ sudo rfkill unblock wifi
[sudo] password for rock:
rock@radxa-zero:~$ sudo rfkill block wifi
rock@radxa-zero:~$ journalctl -f
-- Logs begin at Sat 2023-06-03 21:27:01 UTC. --
Jun 09 19:35:29 radxa-zero prusa.link.serial.serial_queue[2129]: DEBUG: Instruction 'M27 P' enqueued to the front {enqueue_one():323}
Jun 09 19:35:29 radxa-zero prusa.link.serial.serial_adapter[2129]: DEBUG: Sent to printer: b'M27 P\n' {write():335}
Jun 09 19:35:29 radxa-zero prusa.link.serial.serial_adapter[2129]: DEBUG: Printer says: 'Not SD printing' {_read_continually():305}
Jun 09 19:35:29 radxa-zero prusa.link.serial.serial_parser[2129]: DEBUG: Matched RegexPairing for ^(?P<sdn_lfn>/.*\..*)|(?P<no_print>Not SD printing)|(?P<serial_paused>Print saved)|(?P<sd_paused>SD print paused)|(?P<byte_pos>SD printing byte (?P<current>\d+)/(?P<sum>\d+))|(?P<printing_time>(?P<hours>\d+):(?P<minutes>\d{2}))$ with priority 1686339329.6423957 having 1 handler calling {(281473106918128, 281473035541184): <bound method MatchableInstruction.output_captured of Instruction 'M27 P'>} {fire():52}
Jun 09 19:35:29 radxa-zero prusa.link.serial.serial_adapter[2129]: DEBUG: Printer says: 'ok' {_read_continually():305}
Jun 09 19:35:29 radxa-zero prusa.link.serial.serial_parser[2129]: DEBUG: Matched RegexPairing for ^(ok.*)|(Done saving file\.)$ with priority inf having 2 handlers calling {(281473105094832, 281473087518896): <bound method SerialQueue._confirmation_handler of <prusa.link.serial.serial_queue.MonitoredSerialQueue object at 0xffff8f6538b0>>, 281473035476560: <function ThreadedSerialParser.decoupled.<locals>.inner at 0xffff8c4b1e50>} {fire():52}
Jun 09 19:35:29 radxa-zero prusa.link.serial.serial_queue[2129]: DEBUG: Instruction 'M27 P' confirmed {_confirmed():420}
Jun 09 19:35:29 radxa-zero prusa.link.printer_adapter.structures.item_updater[2129]: DEBUG: Value of item print_state has been determined to be PrintState.NOT_SD_PRINTING {set_value():312}
Jun 09 19:35:29 radxa-zero prusa.link.printer_adapter.structures.item_updater[2129]: DEBUG: Scheduling invalidation of item print_state for 1s in the future {schedule_invalidation():354}
Jun 09 19:35:29 radxa-zero prusa.link.printer_adapter.state_manager[2129]: DEBUG: Default expected state change is overridden {wrapper():81}
Jun 09 19:35:29 radxa-zero prusa.link.printer_adapter.telemetry_passer[2129]: DEBUG: Telemetry passer is sleeping... zzz {_update():97}
Jun 09 19:35:29 radxa-zero connect-printer[2129]: DEBUG: Sending Telemetry: <Telemetry:: at 281473036055312> {'state': 'IDLE'} {send():67}
Jun 09 19:35:29 radxa-zero prusa.link.printer_adapter.structures.item_updater[2129]: DEBUG: Item speed_multiplier has been invalidated {invalidate():256}
[... 660 lines cut ...]
Jun 09 19:35:43 radxa-zero prusa.link.serial.serial_parser[2129]: DEBUG: Matched RegexPairing for ^(ok.*)|(Done saving file\.)$ with priority inf having 2 handlers calling {(281473105094832, 281473087518896): <bound method SerialQ [... cut ...]
Jun 09 19:35:43 radxa-zero prusa.link.printer_adapter.state_manager[2129]: DEBUG: Default expected state change is overridden {wrapper():81}
Jun 09 19:35:44 radxa-zero prusa.link.printer_adapter.structures.item_updater[2129]: DEBUG: Item speed_multiplier has been invalidated {invalidate():256}
Jun 09 19:35:44 radxa-zero prusa.link.printer_adapter.structures.item_updater[2129]: DEBUG: Gathering new value for item speed_multiplier {_gather():413}
Jun 09 19:35:44 radxa-zero prusa.link.serial.serial_queue[2129]: DEBUG: Instruction 'M220' enqueued  {enqueue_one():323}

```

TojikCZ, 2023-06-09:
Still failing, or now holding?

neoyarus, 2023-06-09:
failing

neoyarus, 2023-06-09:
after running `systemctl` it looks like the rc.local service failed

`
● rc-local.service                                                                                                                                loaded failed failed
`

TojikCZ, 2023-06-09:
thinking about it seeing the camera twice and not being able to differentiate between them. That might prevent it from getting a picture. I might want to try and incorporate more info from the camera into the camera ID. Usually a camera reports as a single camera with multiple available formats, but this is ali express so it reports itself as two cameras with a single format coz it might have been easier for someone

TojikCZ, 2023-06-09:
you can try running it as root by hand to see the output, if the output is nowhere to be found (rc.local)

neoyarus, 2023-06-09:
ok, running it manually revealed something

rock@radxa-zero:~$ sudo /etc/rc.local
stty: /dev/ttyAMA0: No such file or directory

neoyarus, 2023-06-09:
> thinking about it seeing the camera twice and not being able to differentiate between them

But it works fine on the other device though. Not to mention that there it is still running as a systemd service and getting constantly restarted. That pc is also very old, it has the same amount of ram as the radxa zero, and i think the cpu (some sort of intel atom) is pretty comparable speed-wise.

neoyarus, 2023-06-09:
tried to do a system update, but it looks like this image does not allow that

```
rock@radxa-zero:~$ sudo update-manager
sudo: update-manager: command not found
```

TojikCZ, 2023-06-09:
Sorry, i just thought it would quack like an Ubuntu
Even the kernel is old. 5.10.69
I'm running 6.3.6
Yours is from the September 2021
It's not that bad, but it's def not recent, but it probably won't have much effect anyways
I use arch btw :D so I would try out manjaro, they seem to have a recent image available, but who knows if it works as expected
Again, sorry for not being of much help. I'll be off on the weekend, so see ya on monday

neoyarus, 2023-06-09:
Thanks for the help. I'll probably experiement with different images over the weekend, maybe the majaro ones will support the [overlay](https://wiki.radxa.com/Device-tree-overlays#:~:text=0%2Dgpioao%2D1-,meson%2Dg12a%2Duart%2Dao%2Db%2Don%2Dgpioao%2D2%2Dgpioao%2D3,-Activates%20UART%20AO) that i need for communicating with the printer over gpio pins.
See you on monday :)

neoyarus, 2023-06-17:
It looks like i was able to fix the issue by installing a newer (pre-release) ubuntu [image](https://github.com/radxa-build/radxa-zero/releases/tag/b23). There was a bunch of new issues during install, but once i got them resolved with the help of chatGPT, the camera works fine and prusalink does not crash.

closed 2023-06-17 by neoyarus