prusa3d/Prusa-Link #790: Option to add an IP cam via an RTSP stream

blueone0, 2023-06-15:
It would be great if there is an option to add an local IP camera to prusa link instead of an usb camera, as most camera's support an RTSP stream maybe that is an option to add which should also work with the less power pi zero W

barth11, 2023-07-31:
I'd like to also see this feature. An option to embed IP stream to Dashboard page. Maybe instead of the picture of the print which is currently there.

NateroniPizza, 2026-02-22:
Got my Core One L recently, and while I can understand them not implementing this years ago when people were wanting to add their own DIY cameras, it baffles me why this wasn't immediately implemented when they started selling the cameras.

This is my first Prusa, and the Buddy3D camera implementation on the Core One L feels completely hacked together - why would a new printer released around 2025/2026, that has a camera add-on you can buy with it, not display the camera's view in the printer's web interface by default? It SHOULD just be a hardwired data connection to display in the interface (and not a power-only connection for a separate wifi device you have to hassle with), but barring that then at least do the next most reasonable thing - allow viewing of it in the PrusaLink web interface.