prusa3d/Prusa-Link #795: Uploading files with the same name

neoyarus, 2023-06-21:
Imagine a scenario:

I slice a file and upload it to PrusaConnect from PrusaSlicer.
I print it, see some flaws.
I make small changes to the model, reslice and reupload it. The print time is the same, so the filename is the same too.

Currently, PrusaConnect just won't let me upload a file with the same name. I would have to go into it and delete the file from cloud storage, then from local storage, and only then reupload it. Not only is it very annoying to do, but it is also easy to miss the fact that the file failed to upload, so i can be waiting for the print to start while it is just sitting idle. It is possible to solve this by reconfiguring all my print profiles to add a timestamp(when the file was sliced) at the end, but that makes the names super long, plus its annoying to do.

I think there should be a setting in PrusaConnect for what to do when an uploaded files has the same name as an existing one. The options could be:
1) Keep the original file (that's what it does currently)
2) Overwrite the original file
3) Keep the original file and append "(1)" to the name of the new file (the way windows does)

ondratu, 2023-06-22:
Hi yes, we know about the scenario, and we know how we can solve. At this time PrusaLink has mechanism to check if file exists, and right header to overwrite the file. But we have to do UI modifications which can use that. So this is still in progress now.

cozmo14047, 2023-07-17:
i can upload files with same name now

closed 2023-11-23 by neoyarus