prusa3d/Prusa-Link #806: Add Support for MK4

mikedmor, 2023-07-03:
In its current state, the MK4 is unable to do the following things with the built in PrusaLink:

- Control the printer
- Add/Manage Webcams
- Adjust Settings

I was going to spin up another Raspberry Pi i have laying around to utilized instead of the built in prusalink interface, but it does not currently appear that the MK4 is supported as i cannot get past the configuration Wizard page since it is not detected. Adding support for the MK4 would provide those missing features that the built-in PrusaLink is currently lacking. If these features are missing on the MK4 due to performance reason, an external device makes sense in order to add additional features.

TojikCZ, 2023-07-03:
We'll see. I probably won't get assinged time to work on this. As it's implemented right now, adding support for another firmware that behaves a lot different would not be easy at all, and investing money into an alternative to our own product looks weird.

closed 2023-07-04 by TojikCZ

florianbaer, 2024-01-02:
Seems that these features are now implemented. 
MK4 has a USB-C Port, can I connect a webcam to it and connect it with Prusa-Connect?