prusa3d/Prusa-Link #809: Feature request: PrusaLink 0.7.0 for Prusa Mini+

ivan-sadovsky, 2023-07-05:
It would be beneficial to have a possibility to install the latest PrusaLink (currently 0.7.0) on Original Prusa Mini+ with ESP Wi-Fi module with already installed latest firmware (currently 4.4.1).

TojikCZ, 2023-07-05:
PrusaLink is a linux program that wasn't made to run internally on a microcontroller, so this won't be possible. Your request is coming in right after another that asked for MK4 support. Unfortunately, it has been decided to net pursue adding support fort the buddy printers into the RPi version of PrusaLink. Sorry

ondratu, 2023-07-07:
If you mean, that you want last PrusaLink Web to MINI FW. That is what we are actually doing. New version of PrusaLink Web will be in all Buddy FW releases, with not same functionality, but much more that actual FW for MINI has.

ivan-sadovsky, 2023-07-07:
@ondratu Thanks for clerification. I'm interested in both (i) PrusaLink Web and (ii) PrusaLink API to be able to access api/telemetry from Arduino or RPI.

TojikCZ, 2023-07-07:
Ah, so sorry, i didn't catch that. My bad