prusa3d/Prusa-Link #810: Login on Prusalink from Mac OSX Safari doesn't seem to work

icepick3000, 2023-07-14:
Seems like I am not the only one with this issue;

https://forum.prusa3d.com/forum/introcuctions-instrutctions/problems-logging-in-to-prusa-link-on-mk4/

After entering the password I get the login screen again. 

It works from my windows laptop but not from my iMac with Safari (it does work with Chrome on the iMac).

cozmo14047, 2023-07-17:
same here iphone ipad and imac

seppelicous, 2023-07-22:
Same here, latest MacOS Version, both Safari and Chrome are not able to login.

bobbyh41, 2023-07-29:
Just upgraded my mk3 to a mk4 and I am also having the problem logging in on the iphone in safari, it just kicks me back to the login over and over again.

cozmo14047, 2023-09-13:
will this ever be fixed?

jmckeown2, 2023-12-19:
is this fixed by virtue of buddy firmware 5.1.2?

https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3287

or does that fix need to be duplicated here?