prusa3d/Prusa-Link #811: WiFi disconnects after some time

Redth, 2023-07-17:
Using Pi Zero 2 W, everything works great for a certain amount of time (~5 hours or so).  After this, the device drops off the network and the printer reads "No LAN Access".

I think I've narrowed it down to the wifi adapter's power management putting it to sleep after awhile.

It may be specific to my Unifi setup, not sure, but currently I'm working around it with the advice [found here](https://raspberrypi.stackexchange.com/a/138867):

> Append `/sbin/iwconfig wlan0 power off` to `/etc/rc.local` before the line `exit 0`

It seems that `wireless-power off` in the `/etc/network/interfaces/` file is suggested as well, though I saw reports of mixed success with that only.

Would it be possible to either have an option to turn wireless power management off more easily from the UI, or to have it off by default?

TojikCZ, 2023-08-02:
Looking into this, although I don't think i can measure results, it really seems like it should do what you say it should do, so I see no harm in using it. The implication is, this would be bad if running from a battery, which we are not in the absolute majority of cases.

srinathh, 2023-09-11:
I just ran into this as well & followed the one above. Since RPIs connected to Prusa are always powered, this should be a default setting

TojikCZ, 2023-10-04:
Check if you think this should help once we build a new image #855

tommaso-b, 2026-08-19:
I experience the same problem and finally found a workaround, which works for me.

* My setup:
Prusa I3 MK3S+ with Raspberry Zero 2 and Raspberry camera, sitting in an Ikea Lack enclosure, latest firmware + software

* My problem:
Random WLAN disconnects after several hours of operation.
Obviously Zero 2 keeps on operating as prints do still continue, even if WLAN has stopped.

* Attempts, which did **NOT** work:
  - disabling IPv6 on pi Zero 2
  - disabling WLAN power management on pi Zero 2
  - installing a separate DC buck converter for pi Zero 2 (5.1V, > 3A)
  - installing a modified casing for pi Zero 2 to get larger distance from Einsy board and hence better cooling
  (anyhow pi Zero 2 got never hotter than 63 celsius, even with heat bed running at 90 celsius)

* Final workaround
Set up the watchdog to monitor the WLAN connection.
Restart WLAN on pi Zero 2 in case connection has been lost.
Find below the steps to setup the watchdog accordingly.

1. install watchdog package
sudo apt install watchdog

2. acticate hardware watchdog in file /boot/firmware/config.txt 
dtparam=watchdog=on

3. configure hardware watchdog in /etc/watchdog.conf , edit the settings accordingly :
```
# TB max timeout of 15sec supported by zero 2 hardware
watchdog-device       = /dev/watchdog
watchdog-timeout      = 15

# TB interval for performing tests within timeout period
interval              = 7

# TB instead of resetting the chip we just want to restart the WLAN connection
repair-binary         = /usr/local/bin/network-restart.sh

# TB give more time to handle errors + restart attempts in case of failing ping tests
retry-timeout         = 100
repair-maximum    = 1

# TB a test consists out of 3x pinging the central router of my home network (via WLAN)
# TB any successfull ping is "pass"
ping                 = 192.168.178.1
ping-count          = 3
interface           = wlan0
```

4. create the WLAN restart script /usr/local/bin/network-restart.sh :
```bash
#!/bin/bash
logger "Watchdog: Netzwerk wird neu gestartet (inkl. Wifi)..."

# 1. Wifi-Interface explizit neu starten
if [ -f /sys/class/net/wlan0 ]; then
    sudo ip link set wlan0 down
    sleep 2
    sudo ip link set wlan0 up
fi

# 2. Netzwerkdienste neu starten (falls nötig)
sudo systemctl restart networking

# 3. WPA-Supplicant neu starten (falls Wifi immer noch hängt)
sudo systemctl restart wpa_supplicant

logger "Netzwerk-Neustart abgeschlossen (Wifi sollte jetzt funktionieren)"
```