prusa3d/Prusa-Link #818: [Feature] Ability to add multiple users

CommanderRedYT, 2023-07-29:
We in our hackerspace currently have a Prusa MK3S+. So people know how to use it, we require introductions to each machine (3d printer, electric saws, laser cutter etc.)

For this, it would be really useful if everyone has their own user account. That way we can manage who has got access to the printer.

CommanderRedYT, 2023-08-03:
Hm okay, then I will have to go with sharing the prusalink API key. I can't force them to make a account for a online service (some of them wouldn't like it), and also we need it to work when we have wifi but no uplink to the internet... This will be a bit challenging

CommanderRedYT, 2023-08-03:
I like using connect to monitor it, but people in our hackerspace are i think very against needing to register for something, even if it's free, because of not-selfhosted.

closed 2023-08-07 by ondratu

CommanderRedYT, 2023-08-07:
So what is the solution then? Octoprint?

TojikCZ, 2023-08-07:
That is certainly an option. If you do not like a thing, go with another thing.