prusa3d/Prusa-Link #824: feature request : override passwords or disable security entirely

shukriadams, 2023-08-09:
It would be great if, on Prusalink on the Mk4, we could specify our own passwords, or disable security entirely. For those of us on home networks, behind routers with firewalls, it's not necessary to have long strong passwords on our printers. We understand the risks etc. Many of us are using Prusalink for the convenience of not having to write Gcode to USB keys, it makes no sense to force us to use a password manager for a printer on a network only we have access to.

jmckeown2, 2023-09-10:
Let's be frank. PrusaLink is running unencrypted HTTP. so that nice strong password is being sent to the printer in the clear anyway. So it's already pretty insecure, if the home network is compromised, the printer is exploitable. "Wireshark and wait" WILL compromise the password no matter how strong or weak it is.

bearclaws8, 2023-09-18:
Having the ability to modify the passwords and/or completely disable the need for passwords would be a great enhancement.

I was hoping to install Prusa Link on a completely dead end router setup but haven't been able to because the installation process requires connection to Prusa servers, even for a completely local network only installation.  Which is unfortunate.  I am hoping they can revise the coding around this to permit fully off-line local network setups (see #845 ) as I would like to use the slick looking Prusa Link locally instead of having to rely on octoprint only.

GRIFFCOMMca, 2023-10-24:
I am sure this is open somewhere else, but YES, change password / disable is a must

spontarelliam, 2023-11-19:
Agree, this is an obnoxious requirement in a shared makerspace. The paper printers don't need a password, nor the embroidery machine, or refrigerator, and not even the 3d printers.

cbc02009, 2024-04-02:
Yes, please implement this. I'm behind a firewall and don't need a password.

BoCoKeith, 2024-06-30:
IMHO, the ability to change the password is a must.  It's partly a convenience -- having to find, copy, and paste the password for my MK4 every time I start up PrusaLink is a pain -- but it's also a matter of security -- no one other than me should be managing passwords for the devices on my network.

Just-ForFun, 2024-07-09:
Are they working on this problem?

3d-gussner, 2024-07-09:
This repository is for the Prusa MK2.5/S and MK3/S/+ PrusaLink on SBC or and other Linux based system connected via RPi port or USB to the printer. See https://github.com/prusa3d/Prusa-Link/blob/master/README.md

Please close this issue here an open it in https://github.com/prusa3d/Prusa-Firmware-Buddy as PrusaLink for 32-bit printers like MINI, MK3.5, MK3.9, MK4 and XL run on the controller board.