prusa3d/Prusa-Link #826: Example config is incorrect/out-of-date

kubik369, 2023-08-10:
The printer section in the example config reads:
```
[printer]
; port = /dev/ttyAMA0
; baudrate = 115200
; settings = ./prusa_printer_settings.ini
; mountpoints =
; directories = ./PrusaLink gcodes
```
however, in the code it is read as `directory` https://github.com/prusa3d/Prusa-Link/blob/8b47ea2c52000f82082cee30fbd3624ddbcce0b9/prusa/link/config.py#L177

hence the setting as described in the example config does not work. I can make a PR for this if necessary, it is probably enough to update the example config.

TojikCZ, 2023-08-10:
Yea, the custom mountpoints feature has been broken for quite a while now too, so the example does not need it as well imo, will check

TojikCZ, 2023-09-04:
Fixed in #834

closed 2023-09-04 by TojikCZ