prusa3d/Prusa-Link #829: Prusa Link / MK3S Connection issue when powered up with 3.13.0 firmware

mmoraes80, 2023-08-10:
Hello,

After updating my MK3S MMU2S firmware to 3.13.0 and 3.0.0 respectively, I noticed that, after the printer is powered up, the message "PrusaLink is starting..." stays on the LCD screen.

PrusaLink does seem to load but it does not connect to the printer right after as it keeps showing an error message in PrusaConnect Status on the top left browser window. The full connection is only possible when the printer reset button is pressed, meaning the printer is booting up AFTER PrusaLink has been started.

TojikCZ, 2023-08-10:
Hi, yes, this is a known issue. A temporary workaround would be to disable the mmu during prusalink startup.

The cause:
A new printer type was added - MK3s with an MMU3, I found out because somebodys printer stopped working and they wrote about it on the forum.

An update to mitigate this is coming hopefully soon. 0.7.1. Should be an ota if all goes to plan.

Sorry about the trouble, hope this helps

mmoraes80, 2023-08-10:
> Hi, yes, this is a known issue. A temporary workaround would be to disable the mmu during prusalink startup.
>
> The cause: A new printer type was added - MK3s with an MMU3, I found out because [...]

I am happy to hear that it is a known issue and is being fixed...

TojikCZ, 2023-09-08:
This is solved in the 0.7.1RC1 - a stable release should be out next week. Closing

closed 2023-09-08 by TojikCZ