prusa3d/Prusa-Link #832: Direct G-Code entry for PrusaLink

navara, 2023-08-17:
I want to send individual G-Codes like G28, G0,… to the printer via Prusalink, so I can control it remotely (via WiFi) without having to go to the LCD and do it via the knob.

TojikCZ, 2023-08-17:
I guess you have one of the newer printers. Mini, MK4, XL. Support for them is coming, but no word on when. You won't be able to send commands directly, but we'll expose the basic stuff through buttons and text boxes. Hope that helps.

navara, 2023-08-17:
> I guess you have one of the newer printers. Mini, MK4, XL. Support for them is coming, but no word on when. You won't be able to send commands directly, but we'll expose the basic stuff through buttons and text boxes. Hope that helps.

MK4 :) I'll keep looking for that feature!

Basic movements should be good enough and in case I need more, I'll ask.

sp847242, 2024-03-04:
Something like the console on Octoprint would be real nice. I've used that sometimes for changing internal settings, and then M500 to save it to EEPROM. Or to send it new acceleration limits, like to make the y-axis movement more gentle on a tall print if it's not as stable as I'd expected.

rob-miller, 2024-04-02:
Please consider this a vote for a full console like Duet and apparently Octoprint as mentioned above.  

Right at this moment I would like to do some experimentation with the M300 beep command, and all I can see to do is hardcode a file to try some things.  I haven't tested yet to see if Marlin g-code macros M810-M819 are implemented, but that would be another application.

Seems like users are free to write anything to a gcode file and print it, a console would just make that easier.

ianmackinnon, 2024-11-18:
You can upload and run arbitrary multi-line commands a temporary G-code file using the Prusa Link API.

For example, from a Bash shell (edit the environment variables as necessary):

```bash
PRUSA_API=http://192.168.0.100
PRUSA_API_KEY=0123456789ABCDEF

echo -e "M117 Hello...
G04 S1
M117 World!" \
  | curl -v \
    -X PUT \
    -H "X-Api-Key: ${PRUSA_API_KEY}" \
    -H "Accept: application/json" \
    -H "Content-Type: text/x.gcode" \
    -H "Print-After-Upload: ?1" \
    -H "Overwrite: ?1" \
    -F "file=@-" \
    "${PRUSA_API}/v1/files/local/tmp.gcode"

```

navara, 2024-11-18:
Nice. That would help with an automated test of MMU3 load-in/out of all wires several times, to verify there's no issue; without needing to run and confirm every single load test on the LCD.