prusa3d/Prusa-Link #833: [FEATURE REQUEST] PreHeat Nozzle & Bed while gcode transfers over PrusaLink/PrusaConnect

mrjackson, 2023-08-22:
Is your feature request related to a problem? Please describe.
Larger print files can take a few minutes to transfer.

Describe the solution you'd like
Start the printer preheating as soon as a file is sent with "Upload & Print" from PrusaSlicer.
It could be either a pre-set temperature (based on loaded filament maybe), or actual temperatures from the gcode.
This would minimize the time to start a print.

TojikCZ, 2023-08-22:
Hi, welp, assuming you mean the MINI, MK4 and XL, you're still not at the correct place. However the decision is not to implement this. There are other solutions to this problem being worked on. If everything goes to plan. We will have a better way than just a pre-heat. Thank you for your patience

As for the MK3, the speeds on the Pi are usually sufficient, so we don't consider this necessary, but we are open to feedback

mrjackson, 2023-08-22:
So where is the correct place for this for future reference?

TojikCZ, 2023-08-23:
This would be under prusa3d/Prusa-Firmware-Buddy
But the thing is being worked on regardless, so I wouldn't post more of these as it won't speed it up anyways.