prusa3d/Prusa-Link #844: Wrinting to printer SD

sourekv, 2023-09-16:
Hello,
what could I do to fix / enable wrinting to printer SD (pi card works)?

[image]

sourekv, 2023-09-16:
[image]

TojikCZ, 2023-09-16:
It is not feasible to use gcodes to write to the printer's SD card. Sorry

sourekv, 2023-09-16:
Then the app probably shouldn't look like it can :-(

closed 2023-09-16 by sourekv

TojikCZ, 2023-09-16:
will check if we read the read only flag correctly in the link web ui

sourekv, 2023-09-17:
[image]
In this context make these icons for SD card very little sense.

TojikCZ, 2023-09-22:
Soo just to clarify. What version of PrusaLink do you have?

The PrusaLink Gcodes tab is for files on the raspi sd card. The SD Card tab is for the files on the printer SD card. There should not be an option to upload there visible in the recent versions, if there is, please give us more details