prusa3d/Prusa-Link #845: [security enhancement] Enable installation without sending data to Prusa Connect/servers

bearclaws8, 2023-09-18:
To prevent user data being sent to external servers if the installation is intended to be local network only, or for users without an internet connection on the printers network, modify installation to permit the user to select the specific device instead of connecting to Prusa servers to validate device serial number (as noted in Step 12 [here](https://help.prusa3d.com/guide/prusalink-prusa-connect-with-rpi-3-4-usb-mk2-5-s-mk3-s-_469341)). This will enable a more secure installation for users that do not wish to transmit personal data or have a setup in which they cannot because of a local network that is offline from the internet.

Resolution option, similar to Octoprint installation, is to provide user select-able printers during the installation and then, when the user opts into using Prusa Connect (Step 15 of link above), to then transmit data to Prusa servers then and only then.  This would also aid owners of new Mk4 printers who are more security conscious to limit potential data security concerns as they cannot opt out of having this software installed on their device.

ondratu, 2023-09-18:
How, I'm not sure what are you talking about. PrusaLink for MK3 did not do any connection with Prusa Connect if you really don't want. For example, SN validation is in local PrusaLink code, and you can skip registration printer to Connect, so you can use PrusaLink without internet, and be sure that Prusa really don't know about that.

bearclaws8, 2023-09-18:
I am going based off of the wording Prusa uses in their installation steps, which says:

> PrusaLink verifies your printer's serial number with Prusa servers

Looking at the instructions for the on-board [Pi Zero W installation](https://help.prusa3d.com/guide/prusalink-and-prusa-connect-setup-mk3-s-_221744) it notes the same thing, and several users noted a 500 error, which is a problem connecting to the Prusa servers or validating the SN.

Therefore, the current code does not appear to validate SNs locally and requires connection to the Prusa servers before allowing the user to use it locally only.

TojikCZ, 2023-09-22:
This applies only if the user wishes to use Connect. It is true that we require the serial number even when Connect is not to be used. It would complicate the registering workflow later, although it seems to be kind of an oversight. Changing it now might break things and add extra work. I have verified that the Wizard will work without an internet connection, so you're free to run completely offline without any server pinging

bearclaws8, 2023-09-23:
@TojikCZ Thank you for the clarification.  It is unfortunate it was an oversight, I hope it can be modified in future versions before the software reaches v1.0 (maybe at the same time the feature to create our own password #824 is added :) ).  I'll gather the hardware to set up and test to validate being able to set up the device on a network not internet connected (likely with a pi 3/4 rather than a 2W).  Does this mean that the software/Wizard should no longer give a 500 error or SN check errors like I have seen reported?  I would like to avoid gathering all of the hardware needed only to run into those same issues.

Do you know if the software tries to connect to the Prusa servers, or look for a ping off of the servers at all, while being used without Connect?  I had seen some discussions around printers having issues caused by not being able to find a Prusa Connect server.  I ask because if I want to run this on a non-internet connected network, or one that is sometimes connected, I worry the printer may try to find servers it cannot (or shouldn't) and impact print jobs.  Could correcting code for this issue benefit issues such as #728 or #820 for users looking to use Link only and not Connect?

TojikCZ, 2023-09-23:
You can modify the password in MK3 PrusaLink. The issue tracking itself is a bug at this point. That one refers kinda to the MK4 inability to set your own password and the fact that you cannot disable it on both MK3 and 4, but it's under MK3 PrusaLink issue tracking

bearclaws8, 2023-10-03:
I was able to verify that I could get a printer connected to PrusaLink and successfully upload a file and print on a network completely disconnected from the internet.  The printer showed no errors during setup and had no issues during file upload/printing.  I did not see any obvious calls to external servers with my limited testing and SSH checking.  Although I will try to set up something to monitor the network at a later date.

I did notice that /home/[user]/prusa_printer_settings.ini has a call out to the PrusaConnect server.  Is it possible for the user to comment this text out without causing downstream issues?

> [service::connect]
> hostname = connect.prusa3d.com
> tls = True
> port = 0
> token = 

Although, it does look like this server may be hard coded into some of the PrusaConnect application files too, so commenting it out here may not be of any use.  Also, it looks like PrusaConnect will want to report telemetry data to the Prusa servers once every second.  Is this only while printing or continuously so long as the devices are on?  It's not quite clear how a user can enable/disable Connect and thus the server connections and data sharing.  Would a user be able to start off using the printer with Link and Connect and choose to no longer use Connect and mark a setting such that no telemetry or other data is sent afterwards?  Is this something that can be implemented?