prusa3d/Prusa-Link #858: "Last Modified" Date and Time Inaccurate

bearclaws8, 2023-10-09:
The "Last Modified" date and time when using PrusaLink does not show an accurate date or time.  Currently PrusaLink shows the date and time for a file created today as 5/19/2023 12:01 PM.  This is seen on a non-internet connected network running PrusaLink only, unsure if this issue is also seen when running PrusaConnect.

TojikCZ, 2023-10-09:
You need to either set the time manually, or let the raspberry pi get the time from the network time protocol servers, the same way everything else does it. As a bonus, The Raspberry pi does not have the RTC battery, so your manually set time would break once you turn it off. Only the Raspberry Pi 5 has the RTC battery option

bearclaws8, 2023-10-09:
Thanks for the confirmation. I assumed that was the case based on some other raspberry pi projects. I suppose an RTC hat could help but doesn't seem to be required. I was hopeful that if the system saw no new time information that it could pull a system time from the "Estimated Completion" time it calculated via Prusa Link. But I suppose that's unnecessary code for the majority of people or use cases. 

If you don't think that the system clock being different from a PrusaLink connected computer's clock will cause any issues the you can close this issue out.

TojikCZ, 2023-10-10:
The estimated end time should not be accurate either. No gcode file should contain absolute time, only relative increments. If that's not the case. I would like to know about it even if setting time this way would probably be a weste of it. Thank you

bearclaws8, 2023-10-10:
I posed a shot in #857 from a recent build and the Estimated End time was accurate to my local machine even though the file modified time was wrong. I simply assumed the Estimated End time was calculating based off of my local machine time + print duration.

TojikCZ, 2023-10-10:
Huh, yep, the javascript showing the date is able to use your local time, but unless we used that to synchronise the raspi time before the file was uploaded, we'd have a bad date anyway. Either way, I don't think exploring this is worthwhile. You can set the time or let it set itself using ntp. That feels good enough. Sorry

closed 2023-10-10 by TojikCZ

bearclaws8, 2023-10-10:
Understood, and thank you for following up. Just please, please, _please_ don't change the way Estimated End is calculated because that is honestly the more useful time to have correct per local time.

bearclaws8, 2023-10-15:
Just to note for anyone that checks the closed comments for solutions... Installing an RTC solved the issue and may be an easy solution for anyone looking to run a non-internet connected setup.
[image]

Note, during the installation process you will have to manually toggle the power on your pi as shutdown and reboot commands have not been enabled on the pi using PrusaLink to date (see #853)  You can also set up your RTC on a separate internet connected pi to pull the correct date and time and then install onto your other pi without issue, you should not need to install any new packages onto your PrusaLink pi.