prusa3d/Prusa-Link #861: Focus mode for Pi Camera 3

ingbrzy, 2023-10-15:
Please add ability to adjust auto-focus feature (set focus mode manually) on Pi Camera 3 module... 

Unfortunately in default auto-focus is enabled but it is still out of focus (or focused on wrong area) due to print head fast moving..

Thank you

Useful documentation:
https://www.raspberrypi.com/documentation/computers/camera_software.html

https://github.com/Gordon999/Pi_LIbCamera_GUI

ingbrzy, 2023-10-16:
This is how it looks now on 0.7.1RC1..
[image]

closed 2023-10-21 by ingbrzy

TojikCZ, 2023-10-25:
Sorry for the clunky focusing procedure, but i am happy to see it working