prusa3d/Prusa-Link #870: Cannot add printers in multiinstance (0.7.2)

bulhartv, 2023-10-21:
How to add printers in [multiinstance](https://github.com/prusa3d/Prusa-Link/releases/tag/0.7.2)? I downloaded the image for multiinstance, followed the instructions in the [readme](https://github.com/prusa3d/Prusa-Link/blob/master/MULTIINSTANCE.md) and [Prusa Help Website](https://help.prusa3d.com/guide/prusalink-and-prusa-connect-setup-mk3-s-_221744).
After plugging two MK2.5S via USB into a Rapsberry Pi 3B and running the `prusalink-manager start` command, only one printer got an IP address.
I tried several times to replace the image on the Raspberry Pi, swapping the order of turning on the printers, Raspberry Pi, running the command, but unfortunately nothing helped.

Thank you

seppelicous, 2023-10-23:
May I ask you how started the manager? I copied the new 0.7.2 multiinstaces package on my SD card and the prusalink-manager command is unknown:
`root@prusalink:/home/pi/PrusaLink1# prusalink-manager 
bash: prusalink-manager: command not found`

TojikCZ, 2023-10-23:
Hi @bulhartv can you log in through ssh or a connected peripherals and check lsusb, whether it sees both printers?
When you enter `lsusb -v -d 2c99: | grep iSerial` can you see two different serial numbers?
I assume you have the stock multi-instance image installed.
I do not own any MK2.5S to be able to check hows well they work on multiinstance.
Also, if you could send me a log using the one instance of PrusaLink that does start up -> settings -> syslog i might be able to maybe see something unusual there

TojikCZ, 2023-10-23:
Hi @seppelicous - I cannot replicate what you seem to be experiencing. I flashed the multiinstance image and changed the username in the imager to "pi" same as you did. And it still works as before. What did you do to break it?

seppelicous, 2023-10-23:
I also have issues to add a printer. Here a quote... I tried to wipe all instances with "Clean", but it throws a couple of errors.
After reboot it still seem s to recognise only one printer

> pi@prusalink:~ $ prusalink-manager clean
2023-10-23 14:34:04,116 WARNING {interesting_logger.warning():205} [ThreadPoolExecutor-0_0]: PrusaLink instance not running 
2023-10-23 14:34:19,140 WARNING {interesting_logger.warning():205} [ThreadPoolExecutor-0_1]: Failed to stop - SIGKIL will be used! 
2023-10-23 14:34:19,250 ERROR {interesting_logger.error():215} [MainThread]: Error deleting /home/pi/PrusaLink1/prusalink.pid 
Traceback (most recent call last):
  File "<prusalink-0.7.2>/prusa/link/multi_instance/config_component.py", line 408, in delete_file
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/PrusaLink1/prusalink.pid'
2023-10-23 14:34:19,251 ERROR {interesting_logger.error():215} [MainThread]: Error deleting /home/pi/PrusaLink1/power_panic 
Traceback (most recent call last):
  File "<prusalink-0.7.2>/prusa/link/multi_instance/config_component.py", line 408, in delete_file
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/PrusaLink1/power_panic'
2023-10-23 14:34:19,252 ERROR {interesting_logger.error():215} [MainThread]: Error deleting /home/pi/PrusaLink1/threshold.data 
Traceback (most recent call last):
  File "<prusalink-0.7.2>/prusa/link/multi_instance/config_component.py", line 408, in delete_file
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/PrusaLink1/threshold.data'
2023-10-23 14:34:19,254 ERROR {interesting_logger.error():215} [MainThread]: Error deleting /etc/prusalink/prusalink1.ini 
Traceback (most recent call last):
  File "<prusalink-0.7.2>/prusa/link/multi_instance/config_component.py", line 408, in delete_file
PermissionError: [Errno 13] Permission denied: '/etc/prusalink/prusalink1.ini'
2023-10-23 14:34:19,254 ERROR {interesting_logger.error():215} [MainThread]: Error deleting /etc/udev/rules.d/99-printer1.rules 
Traceback (most recent call last):
  File "<prusalink-0.7.2>/prusa/link/multi_instance/config_component.py", line 408, in delete_file
PermissionError: [Errno 13] Permission denied: '/etc/udev/rules.d/99-printer1.rules'
Failed to send reload request: Permission denied
2023-10-23 14:34:19,280 ERROR {interesting_logger.error():215} [MainThread]: Unhandled exception reached top level 
Traceback (most recent call last):
  File "/home/pi/.local/bin/prusalink-manager", line 8, in <module>
    sys.exit(main())
  File "<prusalink-0.7.2>/prusa/link/multi_instance/__main__.py", line 291, in main
  File "<prusalink-0.7.2>/prusa/link/multi_instance/__main__.py", line 210, in clean
  File "<prusalink-0.7.2>/prusa/link/multi_instance/controller.py", line 66, in remove_all_printers
  File "<prusalink-0.7.2>/prusa/link/multi_instance/config_component.py", line 182, in remove_all_printers
  File "<prusalink-0.7.2>/prusa/link/multi_instance/config_component.py", line 395, in remove_printers
  File "<prusalink-0.7.2>/prusa/link/multi_instance/config_component.py", line 401, in refresh_udev_rules
  File "/usr/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['udevadm', 'control', '--reload']' returned non-zero exit status 1.
pi@prusalink:~ $ prusalink-manager status
usage: prusalink-manager [-h] [-i] [-d] [-u USERNAME] [-p PREPEND_EXECUTABLES_WITH] {start,stop,clean,rescan} ...
prusalink-manager: error: argument command: invalid choice: 'status' (choose from 'start', 'stop', 'clean', 'rescan')
pi@prusalink:~ $ prusalink-manager start
pi@prusalink:~ $ ls
PrusaLink1
pi@prusalink:~ $ ls

TojikCZ, 2023-10-23:
PrusaLink manager NEEDS, HAS TO, MUST to be run as root. Like really, it cannot touch root created files without it. 
If you really really really want to go this route instead of just installing prusalink as root and running `sudo prusalink-manager`, the way to command the manager as root is in /etc/rc.local

`PYTHONPATH=$user_site /home/$username/.local/bin/prusalink-manager -p "PYTHONPATH=$user_site /home/$username/.local/bin/" start`

where user_site is whatever this outputs run as a reguar user `python -m site --user-site`
and username is `id -nu 1000` so "pi" in your case
It was PAIN to figure this out just so the OTA update thing has a chance to work and it sucks, because now it is hard to use for regular people.

Anyway, you now have the exact same problem as bulhartv. Please send me the log file from the one functioning instance. Thank you

seppelicous, 2023-10-23:
I cannot follow you anymore. Please don't understand me wrong, I don't want to use the software different to your original intention. That's why I just flashed it with the original image and tried to follow the help file.

As mentioned "sudo prusalink-manager" is not working at all --> command not found. To conclude I don't know whether it is running or not, I don't see any status nor any feedback if it throws errors or not.

I have only basic experiences with Raspberry Pis, I am more focused on printing. Octoprint is not working with MK4, PrusaLink in Mk4 is slow and the mK4 overall has a bad Wifi reception and does not support any webcam.

Having both printers at one Raspberry Pi, is really a big motivation right now - but I struggle with my Raspbian coding knowledge.

Logfile... There is no log file available in the GUI

TojikCZ, 2023-10-23:
Okay, sorry for assuming your skill level. You mentioned not being able to access the second instance of prusalink, so I assumeb you could access the first one. If you set it up, you should be able to control that one printer right? There should be a settings tab in PrusaLink and at the bottom of it, you should be able to access the log files. Please send a screenshot if that is not the case.

As for controlling the manager, get the root console - `sudo su` then after that, any command you want to run using the manager looks like this for you:
`PYTHONPATH=/home/pi/.local/lib/python3.9/site-packages /home/pi/.local/bin/prusalink-manager -p "PYTHONPATH=/home/pi/.local/lib/python3.9/site-packages /home/pi/.local/bin/" <command>`

so running clean would be: `PYTHONPATH=/home/pi/.local/lib/python3.9/site-packages /home/pi/.local/bin/prusalink-manager -p "PYTHONPATH=/home/pi/.local/lib/python3.9/site-packages /home/pi/.local/bin/" clean`

seppelicous, 2023-10-23:
Thank you for your patience.
Acutally I cannot select any log file. 
[image]

The first, regular is running flawless.
And the given commands are now working 😊

So I cleaned and rescanned, but it still seems to only identify my Mk3, not the MK4.

> pi@prusalink:~ $ sudo su
root@prusalink:/home/pi# PYTHONPATH=/home/pi/.local/lib/python3.9/site-packages /home/pi/.local/bin/prusalink-manager -p "PYTHONPATH=/home/pi/.local/lib/python3.9/site-packages /home/pi/.local/bin/" start
Manager already running
root@prusalink:/home/pi# PYTHONPATH=/home/pi/.local/lib/python3.9/site-packages /home/pi/.local/bin/prusalink-manager -p "PYTHONPATH=/home/pi/.local/lib/python3.9/site-packages /home/pi/.local/bin/" rescan
root@prusalink:/home/pi# PYTHONPATH=/home/pi/.local/lib/python3.9/site-packages /home/pi/.local/bin/prusalink-manager -p "PYTHONPATH=/home/pi/.local/lib/python3.9/site-packages /home/pi/.local/bin/" clean
2023-10-23 17:02:46,252 WARNING {interesting_logger.warning():205} [ThreadPoolExecutor-0_1]: Failed to stop - SIGKIL will be used! 
2023-10-23 17:02:46,255 WARNING {interesting_logger.warning():205} [ThreadPoolExecutor-0_2]: Failed to stop - SIGKIL will be used! 
2023-10-23 17:02:46,367 ERROR {interesting_logger.error():215} [MainThread]: Error deleting /home/pi/PrusaLink1/power_panic 
Traceback (most recent call last):
  File "<prusalink-0.7.2>/prusa/link/multi_instance/config_component.py", line 408, in delete_file
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/PrusaLink1/power_panic'
2023-10-23 17:02:46,369 ERROR {interesting_logger.error():215} [MainThread]: Error deleting /home/pi/PrusaLink1/threshold.data 
Traceback (most recent call last):
  File "<prusalink-0.7.2>/prusa/link/multi_instance/config_component.py", line 408, in delete_file
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/PrusaLink1/threshold.data'
root@prusalink:/home/pi# PYTHONPATH=/home/pi/.local/lib/python3.9/site-packages /home/pi/.local/bin/prusalink-manager -p "PYTHONPATH=/home/pi/.local/lib/python3.9/site-packages /home/pi/.local/bin/" start
root@prusalink:/home/pi# PYTHONPATH=/home/pi/.local/lib/python3.9/site-packages /home/pi/.local/bin/prusalink-manager -p "PYTHONPATH=/home/pi/.local/lib/python3.9/site-packages /home/pi/.local/bin/" rescan
root@prusalink:/home/pi# ls
PrusaLink1
root@prusalink:/home/pi#

TojikCZ, 2023-10-23:
So it just does not let you select anything? Empty dropdown? If you clean, you need to start it again, the re-scan just prompts the running instance to re-scan, if there is none running, nothing happens. Also, that should happen automatically upon a printer being connected.

Also, you haven't told me what this outputs - `lsusb -v -d 2c99: | grep iSerial`
if in ssh, you can have a look in /va/log/syslog. You can use scp to get the file off of the raspi into your pc.

`scp pi@<your_ip>:/var/log/syslog .` so if it were runing on 192.168.1.2 it would go like this: `scp pi@192.168.1.2:/var/log/syslog .`

bulhartv, 2023-10-23:
> Hi @bulhartv can you log in through ssh or a connected peripherals and check lsusb, whether it sees both printers? When you enter `lsusb -v -d 2c99: | grep iSerial` can you see two different serial [...]

Hi @TojikCZ, I followed your advices and there are my results. I did clean installation of the multiinstance image. Connected both printers and started all up.

The first I checked I see connected priners
```
lsusb
Bus 001 Device 005: ID 2c99:0001 Prusa i3 MK2S
Bus 001 Device 004: ID 2c99:0001 Prusa i3 MK2S
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Tried your command, I just got SN of the printer which is connected (in Prusa Link a got IP), the other one has some unknown characters
```
lsusb -v -d 2c99: | grep iSerial
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing
  iSerial                 3 ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
  iSerial                 3 CORRECT SERIAL NUMBER
```

Printer 1 got IP, so I did setted up Prusa Link. Than I followed your advices with root and start of prusa-manager and got info: „Manager already running“.

Then I tried to reboot Rpi and tried command ```rescan```, but nothing happend with the Printer 2

[syslog.txt](https://github.com/prusa3d/Prusa-Link/files/13078829/syslog.txt)
_I haven't seen it at the first, but the problem was, that I used Safari. When I opened PrusaLink in Chrome, I see all logs_

Thanks for your patience and time!

TojikCZ, 2023-10-23:
Thank you, that was an absolutely awesome writeup, also, here we go, that serial number is the problem. I grab it and check if it's in a format we support, but some printers weren't so lucky and didn't get one.
Are you comfortable connecting to serial through anything other than pronterface? I am not sure, but i think we might be able to flash your serial number if we can communicate with the chip that holds it.

If you can, try sending `;C32u2_FWV` to the printer without the S/N. If it answers something like 1.3 we're in business

bulhartv, 2023-10-24:
Alright, thank you for your fast response.

What do you exactly mean by „connecting to serial through anything other than Pronterface“? Just point me in the direction of what method I should use (or what software)?

seppelicous, 2023-10-24:
> okay, there is your problem, raspi PrusaLink is not written for the MK4 at all, let that one run on its own and use raspi just for the MK3 in single instance mode

Oh okay, that explains a lot - I was not aware of the fact that the MK4 might not be supported. That is very sad, as the internal PrusaLink does not support USB cameras. And the Mk4 ESP has a ver bad signal strength, compared to the Raspi next to it.
**Is there any plan to support the Mk4 in the feature?** There is also another advantage... In our company we have some Mk3 and Mk4s and all are planned to bring them into our network. But it is easier to do it through one Raspi then having the Mk4s as separated network devices. 
Anyway, thank you very much for your quick response!!

TojikCZ, 2023-10-24:
a serial terminal in chrome https://www.serialterminal.com/

bulhartv, 2023-10-24:
Thanks, I followed your instructions and got this result for `;C32u2_FWV`. There is probably a problem with the board. I bought it in March 2021 from Prusa online store. It is a pitty that it isn't under warranty any more.

```
start
echo: 3.13.1-6876
SpoolJoin is Off
echo: Last Updated: Aug 31 2023 10:25:57 | Author: (none, default config)
echo: Free Memory: 2809  PlannerBufferBytes: 1760
echo:Stored settings retrieved
adc_init
Sending 0xFF
echo:SD card ok
> ;C32u2_FWV
Command not found!
```

I tried read command of SN and got this
```
> ;C32u2_SNR
S/N is:�������������������
```

Do you have any other idea?

TojikCZ, 2023-10-24:
That seems fine, now to try and write it. Ideally, figure out how pasting works in the serial console, prepare your ;C32u2_SNW command and once absolutely sure it is correct, paste it in, no newlines needed.

Also weird you cannot get the FWV to work, did you retry it?

bulhartv, 2023-10-24:
Alright, but I got a small problem. Because this MK2.5S printer is upgraded from MK0 2.85, there is no serial number on the frame. Can I find serial number somewhere on the Rambo?

TojikCZ, 2023-10-24:
Well, that is unfortunate. Two options, plead the official support to obtain one for you. 

Or and that is absolutely not the thing you are supposed to do - make one up. just make the last two numbers of the first four number group some weird year like 42 or something that is not likely to occur on any printer. Maybe do the year this printer was made in 😁
Also make sure to have the S/N letters in the exact same places

bulhartv, 2023-10-28:
Thanks, I contacted support on Thursday but I am still waiting for their response. I will let you know.

TojikCZ, 2023-10-28:
Yea, i fo not think that's gonna succeed, but it's worth giving it a shot

TojikCZ, 2023-11-02:
Hi, any updates? Can we close this? I cannot allow printers without S/N through either way, so there's not much utility in this anymore. I am still open to providin assistance with setting the S/N if you need

bulhartv, 2023-11-02:
Hi, I got a reply three days ago telling me to try connecting the printer to my computer. So I replied that in conjunction with you, I have already tried. I still have to wait for a while, and he will ask his colleagues if they can trace the serial number.

bulhartv, 2023-11-03:
@TojikCZ They couldn't trace the serial number. So the only option is to enter the new serial number into the printer. I'd be grateful if you could help me with that. What should be the exact S/N structure?

TojikCZ, 2023-11-03:
CZPX0711X004XC00042

I believe the numbers are like so:
First two - week of year
second two - year
I'd put the week and year of your MK0 coming alive for nostalgia
0004 would be the product code? 0004 MK3S+ so put all zeroes i guess? or leave 0004 for up to date look or make up your own
The last five are what keep the uniqueness within that window of time, put memes or whatever 
07734 upside down spells HELLO
IDK if C stands for assembled and what letter is used for kits, the guide to the logic of these was pulled down and i cannot find it, so going off memory here. Just pick something that shouldn't collide with any and you'll be fine

The regex that checks it looks like this
`VALID_SN_REGEX = re.compile(r"^(?P<sn>^CZPX\d{4}X\d{3}X.\d{5})$")`

bulhartv, 2023-11-03:
Thanks, but I messed up..I have no idea, why is the C missing...

`> ;C32u2_FWV`
`Command not found!`

`> ;C32u2_SNR`
`S/N is:�������������������`

`> ;C32u2_SNWCZPX0115X000XV07734`
`S/N burned:ZPX0115X000XV07734`

`> ;C32u2_SNR`
`S/N is:ZPX0115X000XV07734`

TojikCZ, 2023-11-03:
crap, it had been expecting a : welp, that was not docummented properly 😑

TojikCZ, 2023-11-03:
can it overwrite it? I do not expect that to be the case

bulhartv, 2023-11-03:
No, I can not.

`> ;C32u2_SNR`
`S/N is:ZPX0115X000XV07734`
`> ;C32u2_SNWCZPX1914X000XG61826`
`S/N burn FAILED!`

TojikCZ, 2023-11-03:
Welp, sorry about that, there might be a solution but it will take a considerably longer time.

The only way it would work for now would be to edit the valid sn regexp in the code so it matches for you, make the first C optional by putting a questionmark behind it. But you would have to re-install from source.

You would download the latest release tar, unpack it and transfer it to the pi, 
edit the regex found in Prusa-Link/prusa/link/printer_adapter/structures/regular_expressions.py and then install that with pip as a regular user
pip install .

to transfer, you can use scp command line tool or filezilla.
But then it would break on update

Soo either that or i'll ask around. I got a word about another solution being maybe available later

bulhartv, 2023-11-24:
I'll eventually have a new rambo board, hopefully multiinstance will work with no issues. Thanks a lot for your time a patience.

TojikCZ, 2023-11-27:
Hi, I know, discovered we are able to send one with minimal hassle, so I asked around and it panned out. Sorry for the added work. I seriously hope this one will work out of the box. If not, at least I now know the command to write the S/N needs a colon before the CZPX part.