prusa3d/Prusa-Link #871: prusalink-manager: command not found

seppelicous, 2023-10-23:
Actions I did:
1. Downloaded and flashed "prusalink-multi-instance-0.7.2.img.xz" with the Raspberry Pi flasher. Manually activated SSH and changed username and password (pi)
2. Inserted SD card in Raspi 4
3. SSH into Raspi:

`Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
pi@prusalink:~ $ sudo -s
root@prusalink:/home/pi# prusalink-manager start
bash: prusalink-manager: command not found
root@prusalink:/home/pi# 
`

What am I doing wrong?

I appreciate your help

seppelicous, 2023-10-23:
Update:
I can run the command with user pi, but I get no feedback.

> > pi@prusalink:~ $ prusalink-manager -i
> usage: prusalink-manager [-h] [-i] [-d] [-u USERNAME] [-p PREPEND_EXECUTABLES_WITH] {start,stop,clean,rescan} ...
> 
> Multi-instance suite for PrusaLink
> 
> positional arguments:
>   {start,stop,clean,rescan}
>                         Available commands
>     start               Start the instance managing daemon (needs root privileges)
>     stop                Stop any manager daemon running (needs root privileges)
>     clean               Danger! cleans all PrusaLink multi-instance configuration
>     rescan              Notify the daemon a printer has been connected
> 
> optional arguments:
>   -h, --help            show this help message and exit
>   -i, --info            include log messages up to the INFO level
>   -d, --debug           include log messages up to the INFO level
>   -u USERNAME, --username USERNAME
>                         Which users to use for running and storing everything
>   -p PREPEND_EXECUTABLES_WITH, --prepend-executables-with PREPEND_EXECUTABLES_WITH
>                         Environment variables and path to the executables directory
> pi@prusalink:~ $ sudo prusalink-manager -i
> sudo: prusalink-manager: command not found

TojikCZ, 2023-10-23:
Yep. So your image is behaving correctly in this regard. Nice.
The manager is autostarted upon bootup automatically. I had a great deal of difficulty making it be installed under the user jo, but running under root. You can see my pain in /etc/rc.local, that's where the start command used to run the manager is. If you want to run it manually, please just install it under root and run it as root, it is simpler, but the ota updater breaks for sure

seppelicous, 2023-10-23:
@TojikCZ Thank you! Then I misunderstood the help file, where I am told to start the manager with root. But great, that everything should work now...

Nevertheless I have no idea how to access the second instance of Prusa-Link. 
Both Purses are connected, but what shall I do now?

TojikCZ, 2023-10-23:
Yes, the multi-instance starts as root, it needs to run as root. That in itself doesn't mean that I can get it to global $PATH without root privileges

I cannot test it with only one printer at home, so now you actually are trying to solve the same issue as the guy in the previous issue you posted in.

Why are you at three different places at once?

seppelicous, 2023-10-23:
Thank you, the original issue "command not found" can be closed now. The second issue I mention here is already opened, so I will move to that one.
Thank you for your help!

closed 2023-10-23 by seppelicous