prusa3d/Prusa-Link #890: Printer goes offline after few minutes after start

ingbrzy, 2023-11-21:
Hello.. 

My setup:

- Printer MK3S+
- FW 3.13.2-7080
- Prusa-link 0.7.2
- Prusa Connect SDK 0.7.1
- RPi Zero W2
- RPi Camera 3

Printer goes offline after few minutes.. logs in attachment.. 
Already tried to clean install but got same issue..
Can you help me solve this issue?

I see a lot of errors in the logs like this:

`Nov 20 14:41:58 prusa01 connect-printer[536]: ERROR: HTTPSConnectionPool(host='connect.prusa3d.com', port=443): Max retries exceeded with url: /p/telemetry (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x71349af0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')) {loop_step():744}
Nov 20 14:42:23 prusa01 camera_controller[536]: ERROR: Unexpected exception caught in SDK snapshot loop! {snapshot_loop():186}#012Traceback (most recent call last):#012  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 169, in _new_conn#012    conn = connection.create_connection(#012  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 73, in create_connection#012    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):#012  File "/usr/lib/python3.9/socket.py", line 953, in getaddrinfo#012    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):#012socket.gaierror: [Errno -3] Temporary failure in name resolution#012#012During handling of the above exception, another exception occurred:#012#012Traceback (most recent call last):#012  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 699, in urlopen#012    httplib_response = self._make_request(#012  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 382, in _make_request#012    self._validate_conn(conn)#012  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 1012, in _validate_conn#012    conn.connect()#012  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 353, in connect#012    conn = self._new_conn()#012  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 181, in _new_conn#012    raise NewConnectionError(#012urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x71354d48>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution#012#012During handling of the above exception, another exception occurred:#012#012Traceback (most recent call last):#012  File "/home/jo/.local/lib/python3.9/site-packages/requests/adapters.py", line 486, in send#012    resp = conn.urlopen(#012  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 755, in urlopen#012    retries = retries.increment(#012  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 574, in increment#012    raise MaxRetryError(_pool, url, error or ResponseError(cause))#012urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='connect.prusa3d.com', port=443): Max retries exceeded with url: /c/snapshot (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x71354d48>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))#012#012During handling of the above exception, another exception occurred:#012#012Traceback (most recent call last):#012  File "/home/jo/.local/lib/python3.9/site-packages/prusa/connect/printer/camera_controller.py", line 174, in snapshot_loop#012    res = item.send(self.session, self.server)#012  File "/home/jo/.local/lib/python3.9/site-packages/prusa/connect/printer/camera.py", line 70, in send#012    res = conn.request(method=self.method,#012  File "/home/jo/.local/lib/python3.9/site-packages/requests/sessions.py", line 589, in request#012    resp = self.send(prep, **send_kwargs)#012  File "/home/jo/.local/lib/python3.9/site-packages/requests/sessions.py", line 703, in send#012    r = adapter.send(request, **kwargs)#012  File "/home/jo/.local/lib/python3.9/site-packages/requests/adapters.py", line 519, in send#012    raise ConnectionError(e, request=request)#012requests.exceptions.ConnectionError: HTTPSConnectionPool(host='connect.prusa3d.com', port=443): Max retries exceeded with url: /c/snapshot (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x71354d48>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
`

[syslog-mk3.txt](https://github.com/prusa3d/Prusa-Link/files/13423330/syslog-mk3.txt)
[messages-mk3.txt](https://github.com/prusa3d/Prusa-Link/files/13423332/messages-mk3.txt)

TojikCZ, 2023-11-21:
Hi, looks like your pi has trouble talking to your wireless card

I gave a few of the messages to chatgpt as i did not know what the kernel trace was and these are the things it came up with

> Hardware Address 00:00:00:00:00:00: The log repeatedly shows dhcpcd reporting that wlan0 with a hardware (MAC) address of all zeroes (00:00:00:00:00:00) is claiming an IP address (10.192.35.20). This is unusual because a MAC address of all zeroes is typically invalid. It might indicate a problem with your network interface card (NIC) or its driver. In some cases, it could also be a sign of a misconfigured network or a network security issue, like MAC address spoofing.

> Kernel Failures and SDIO (Secure Digital Input Output) Errors: The kernel logs (brcmfmac entries) are showing repeated failures related to SDIO, which is often used for interfacing with wireless network cards on embedded systems like Raspberry Pi. Errors like brcmf_sdio_readframes: RXHEADER FAILED: -110, brcmf_sdio_rxfail, and failed backplane access suggest there are problems with the SDIO communication, possibly due to hardware issues, driver problems, or even power supply inadequacies.

ingbrzy, 2023-11-21:
Wireless card is integrated into the Rpi.. so it means, Rpi is probably bad and has to be replaced?

TojikCZ, 2023-11-21:
It also suggested power supply issues, if you have the pi connected to the gpio pins, please try using an OTG cable and an external power source. If the issues go away, it was just the stability of the 5V line in the printer. If using an external powersource, try another one if possible. You can also try using an external ethernet or wifi card before tossing the raspi. But yes, you are getting the right idea here, unfortunate as it might be.

ingbrzy, 2023-11-21:
Thank you, yea it is connected via GIO pins..
Can I just plug power supply to Rpi and left it connected to GPIO at same time?

TojikCZ, 2023-11-21:
Please do not do that, when you turn of your printer, the pi will provide power to it and something might burn. I tried it and the printer actually boots up all hissing and unhappy the pi straining to keep it alive

ingbrzy, 2024-02-01:
Mysteriously solved by itself since 25.01.2024

closed 2024-02-01 by ingbrzy

TojikCZ, 2024-02-01:
just a note, you could snip the 5v line on the pi, I think it's the top one, that way you can power the pi externally while using the pins and stuff should be fine