prusa3d/Prusa-Link #899: Prusa Link doesn't preview QOI images in the web interface.

Claghorn, 2023-12-05:
I changed the slicer setting from PNG to QOI in order to get thumbnail previews on my MK4 LCD screen (which did work), but now if I connect to PrusaLink with google chrome the web interface doesn't display the thumbnail.

wesseli01, 2023-12-08:
I got the same Problem.

Claghorn, 2023-12-08:
I took a look at the latest configuration for Prusa Slicer and I see the "official" G-code thumbnails setting is
`16x16/QOI, 313x173/QOI, 440x240/QOI, 480x240/QOI, 640x480/PNG` which seems to give me thumbnails everywhere again.

TojikCZ, 2024-01-02:
That is correct. PrusaLink will not do QOI. Just leave a PNG in the slicer. QOI is a RAM saving format for the 32 bit anyways, in MK3 PrusaLink it has no advantages that I am aware of. If you are talking about the MK4/XL PrusaLink, then that belongs to the prusa3d/Prusa-Firmware-Buddy repo

JamesRHarris, 2024-02-12:
This appears to be a dupe of https://github.com/prusa3d/Prusa-Link-Web/issues/475

TojikCZ, 2024-02-13:
Well prusa3d/PrusaLink does not support the newer buddy based printers at all.

So yea, it's a duplicatein a wrong repository. Thanks closing