prusa3d/Prusa-Link #919: Printing Time not accurate ~23 hours into a 36 hour print

Braddg, 2023-12-22:
Prusa XL 5T Printing Time not accurate.  It has been running for 23 hours at this point, but says "Less than a minute" under Printing Time.  I'm connected directly to the Prusa XL from my PC on a local network.  The Last Modified, Remaining time and Estimated end time all appear as expected. Prusa XL firmware 5.1.2

[image]

Braddg, 2023-12-22:
Just realized this should go under Prusa Link Web instead.  Closing

closed 2023-12-22 by Braddg

TojikCZ, 2023-12-22:
Hi, it's probably prusa firmware buddy, i would move it, I just am in a holiday lazy mode today so I forgot