prusa3d/Prusa-Link #931: Raspberry Pi 5 Support

derekakelly, 2024-01-23:
Hello,
I can't seem to find any reference to the new Rapsberry PI 5. Is it supported? Would I just follow the steps for the 3/4 to set it up?

Thanks!

TojikCZ, 2024-01-24:
Hi, Raspi 5 is not supported, the base image for 0.7.2 has been created with the old base rpi image. 0.8.0 is on the new base but I need to test it. I do not have the 5. I'm still waiting for my order to arrive.

derekakelly, 2024-02-05:
I'd be happy to test it out whenever it's ready. I just got a Raspberry Pi 5 and I'm printing with an MK3S+ MMU3.

TojikCZ, 2024-02-05:
I think that doing sudo rpi-update over ssh could make it work, but since the combo of new rpi firmware and kernel with older debian might not have been tested by rpi foundation, it's at your own risk. If you do update debian as well, you might need to install python-picamera from apt and maybe do some other stuff too. At this point I have tested the rPi 5 and with new debian and firmware, stuff seems to be working. The only problem being that they broke the noninteractive mode in raspi-config, so I'll be re-doing that part of the image-builder script

sportifia27, 2024-03-26:
Hi guys, i have the same problem, i have a raspberry pi 5 and prusalink 0.72 doesn't work, how can I solve the problem? (I'm not a advanced user with raspberry)