prusa3d/Prusa-Link #933: MultipleInstances not working for second printer

UlitmateCro, 2024-01-25:
Hey guys, looking for a little bit of help, not sure if this is the place to do it or not. I have managed to use and getting the manager running by using  prusalink-manager start in the root of the RPI4, however nothing happens from here, It still only gives the first printer connected an IP, am I doing something wrong here ?

TojikCZ, 2024-01-26:
Hi, sorry, not entirely confident with the validity of info here, so lets go over this one by one.
Implied -> You have not used the multiinstance image
Implied -> You are running prusalink-manager under the user root (with sudo)

What version of PrusaLink are you running? My colleague told me you're on 0.7.1, that was a non-release as it was broken. 0.7.2 is what you need

UlitmateCro, 2024-01-27:
Hey, we figured it out, I thought I applied the multi instance file to the
pi but turns out I didn’t, all sorted and is working now.

Thanks for the quick response though

Kind regards
Deen

TojikCZ, 2024-01-27:
Awesome, glad it's working

closed 2024-01-27 by TojikCZ