prusa3d/Prusa-Link #943: systemd unit for prusalink

naikymen, 2024-03-04:
It would be nice to complete the "pip" installation of prusalink with a few more things, including:

- Installing prusalink from a virtual environment. Otherwise pip install just fails nowadays, because of the very much welcome "externally managed environments".
- A systemd unit that starts prusalink from that virtual environment during boot.

I've made a unit and it seems to be working fine: https://gist.github.com/naikymen/7e1d1bd23ae0709b0b72b845b480f57a

I chose to enable it as a user unit. The `--foreground` option was convenient. :)

Thanks for the software!

TojikCZ, 2024-03-04:
Yep, I was denied making the systemd service. Because "some systems don't use it", so we have it the old way

nvtkaszpir, 2024-09-21:
You don't have to add a code to install it, but at least it could be in the repo for others as reference in the docs.

naikymen, 2024-09-21:
> it could be in the repo for others as reference in the docs.

Certainly! I think it would be welcome as an option by some.