prusa3d/Prusa-Link #953: BUG: XL Printer goes offline when extruder heat command is send

ingbrzy, 2024-04-08:
Printer: Prusa XL 5 heads
Firmware: 5.1.2

Nozzle temperature is shown as 6°C when all heads are parked
[image]

When all heads are parked and I send command via PrusaConnect Control to preheat NOZZLE (any temp), printer goes instantly OFFLINE and error is shown on the screen

[image]

ingbrzy, 2024-04-11:
one more point, when all heads are parked, nozzle position overlaps with text on Prusa Connect..

[image]

ingbrzy, 2024-04-17:
solved by 6.0.0 RC3 Firmware

ingbrzy, 2024-04-17:
sorry, not fixed.. just tested again..

ingbrzy, 2024-04-17:
where to report XL Prusa Connect issues?

TojikCZ, 2024-04-17:
It's like multiple issues at once, one or two for XL firmware and some for connect. I cannot even move this issue to the appropriate repo since it's at least 3 in 1.

Again, this repo deals with the MK3 PrusaLink raspi thing,

Yes i'll go and split it myself :/ but please, use the connect feedback thing on the bottom right for connect issues and Prusa-Firmware-Buddy for FW ones