prusa3d/Prusa-Link #955: Live Raspberry CPU temperature monitoring and serial transmission console on PrusaLink web page.

dwprod38, 2024-04-10:
After installing prusalink 0.7.2 on Raspberry Pi Zero 2 itself plugged on Prusa MK2.5S+ with serial USB cable, I've noticed a layer shift phenomenon after ~2hours printing on 3 different STL files... After changing USB cable: same issue...

After transferring identical STL files to SD card (downloaded files from prusalink on PC and copied to SD card to check corrupted files or not): everything works fine after ~4 hours printing on MK2.5S+...

Moving this Raspberry Pi Zero 2 to MK3S+ (plugged on Einsy Board) and used same files :  everything works fine after ~4 hours printing...

So I've suspected overtemp issue on CPU of Raspberry Pi Zero 2 or data corrupted on serial USB responsible of layer shift. For investigation, I need to visualize CPU temperature and serial transmission checksum during printing.

Thanks for support ;)

TojikCZ, 2024-04-10:
Hi, sorry, there's no temp monitoring for the pi. You'd have to set up that yourself. However, I use checksums for the print gcode. Not sure if the pi produces log messages for overtemp into the syslog. You can download it in PrusaLink web, umder settings tap. You can send that to me if you want. Overtemp should not cause layer shifts tho. The gcode should still transfer correctly, it would just get slowed down, if it slows down too much, you would see visible print quality issues like blobs, as the nozzle stops in place waiting for the next gcode line
. You should not get layer shifts from that. This is interesting, but unfortunately not much i will be able to do about it 🫤 unfortunately MK3 PrusaLink is reeeeally low priority now.

dwprod38, 2024-04-10:
Thx for answering!

A simple command line could be used in prusalink --> watch -n 1 vcgencmd measure_temp for displaying live CPU monitoring ;)

Back to MK2.5S+ and RPi Zero 2 issue: everything works fine with Mk3S+ (serial communication used with GPio port) but something gone wrong with micro USB serial port... I'll try with slow speed printing set at 80% to observe results.

And if I have time, I'll try to use old RPi3 connected to MK2.5S+ with serial USB port...

See you !

TojikCZ, 2024-04-10:
Yea, you can use ssh to log into the raspi. Setup is the same as with usual raspi os. either you enable SSH in the imager, or you enable it by putting an empty file named "ssh" into the bootfs partition of the SD card. After booting, you can connect to the raspi using the login credentials you specified, or the default ones user: `jo` password: `raspberry`