prusa3d/Prusa-Link #956: Report MK4 board temperature in telemetry data

Zappes, 2024-04-12:
As the **owner of a Prusa Enclosure**
I want to **monitor the xBuddy board temperature**
So that **I can take measures when the temperature gets too high**

**Background**

When printing high temperature filaments like PC Blend or Nylon in the enclosure, the temperature of the mainboard can get into ranges where proper functioning isn't guranateed anymore. I would like to be able to react to such a situation, e.g. by activating a cooling solution.

My ultimate goal is to control the behavior of my system using Home Assistant. The data source for that system is the [PrusaLInk Integration](https://www.home-assistant.io/integrations/prusalink) which, in turn, is based on the telemetry data from PrusaLink. For that reason, the first step towards my goal would be the inclusion of the board temperature in the telemetry data.

3d-gussner, 2024-05-15:
@Zappes This is the repository for the external SBC/computer PrusaLink. For the MINI, MK3.5, MK3.9, MK4 and XL PrusaLink is running on the controller board. Please open MK4 related issues [here](https://github.com/prusa3d/Prusa-Firmware-Buddy)

3d-gussner, 2024-07-05:
@Prusa-Support Please close this issue

Zappes, 2024-07-05:
Sorry, the mail notification for your first reply was withheld from me by my junk filter... Closing. :)

closed 2024-07-05 by Zappes

3d-gussner, 2024-07-05:
@Zappes Thanks, very appreciated. Closing issues helps to keep track of the issues.